ghdl/ghdl-yosys-plugin #1: Linking error when building yosys with GHDL support

Obijuan, 2017-02-10:
I've follow the instructions to build ghdlsynth. In the final step, when building yosys I got this link error:

```
$ make
[100%] Building yosys
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `__gnat_install_SEH_handler'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `system__restrictions__abort_allowed'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `system__stack_usage__is_enabled'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `ada__calendar__delays__to_duration'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `__gnat_stack_usage_output_results'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `system__parameters__adjust_storage_size'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `system__multiprocessors__number_of_cpus'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `system__stack_usage__compute_result'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `system__stack_checking__operations__invalidate_stack_cache'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `_ada_ada__exceptions__is_null_occurrence'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `system__stack_usage__fill_stack'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `system__stack_usage__report_result'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `system__restrictions__run_time_restrictions'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `__gnat_stack_usage_results'
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../x86_64-linux-gnu/libgnarl-5.so: undefined reference to `system__stack_usage__initialize_analyzer'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
Makefile:379: recipe for target 'yosys' failed
make: *** [yosys] Error 1
```

Do you have any clue on what's not going ok?  I've successfully build ghdl and the libghdlsynth

I am testing it in ubuntu 16.04 64-bits

Thanks a lot for you great contribution! There are plenty of people in the community eager to use ghdl for synthesizing. The union between ghdl and yosys is really great! Thanks!  :-)

tgingold, 2017-02-10:
On 10/02/17 20:29, Juan Gonzalez-Gomez wrote:
> I've follow the instructions to build ghdlsynth. In the final step, when
> building yosys I got this link error:
>
> |$ make [100%] Building yosys [...]

That's weird, but I will try to reproduce.

libghdlsynth must have been linked with libgnat-5.so

> I am testing it in ubuntu 16.04 64-bits
>
> Thanks a lot for you great contribution! There are plenty of people in
> the community eager to use ghdl for synthesizing. The union between ghdl
> and yosys is really great! Thanks! :-)

Do not expect too much now, as this is very WIP!

Tristan.

Obijuan, 2017-02-10:
> Do not expect too much now, as this is very WIP!

I know I kow :-)   I just want to test it and report bugs for improving it

tgingold, 2017-02-10:
On 10/02/17 20:29, Juan Gonzalez-Gomez wrote:
> I've follow the instructions to build ghdlsynth. In the final step, when
> building yosys I got this link error:

[...]

> Do you have any clue on what's not going ok? I've successfully build
> ghdl and the libghdlsynth
>
> I am testing it in ubuntu 16.04 64-bits

I have just added a Makefile, so that it could be built as a module.
Should be easier.

I have tested it on ubuntu 14

Tristan.

juanmard, 2017-02-10:
Same errors with libgnarl-6.so  :(
```
$ make
[100%] Building yosys
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `system__stack_usage__fill_stack' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `system__stack_usage__is_enabled' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `system__stack_usage__initialize_analyzer' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `system__multiprocessors__number_of_cpus' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `__gnat_install_SEH_handler' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `system__parameters__adjust_storage_size' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `_ada_ada__exceptions__is_null_occurrence' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `system__stack_checking__operations__invalidate_stack_cache' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `system__restrictions__run_time_restrictions' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `ada__calendar__delays__to_duration' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `system__restrictions__abort_allowed' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `system__stack_usage__compute_result' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `__gnat_stack_usage_output_results' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `__gnat_stack_usage_results' sin definir
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgnarl-6.so: referencia a `system__stack_usage__report_result' sin definir
collect2: error: ld returned 1 exit status
Makefile:377: fallo en las instrucciones para el objetivo 'yosys'
make: *** [yosys] Error 1
$ 

```

Obijuan, 2017-02-10:
I've tested the new Makefile. It seem that there is still a problem with how the libghdlsynth library is compiled

```sh
$ make GHDL_PREFIX=/usr/local
yosys-config --exec --cxx --cxxflags --ldflags -o ghdl.so -shared ghdl/ghdl.cc -DYOSYS_ENABLE_GHDL -I/usr/local/include /usr/local/lib/libghdlsynth.so -Wl,-rpath,/usr/local/lib --ldlibs
/usr/bin/ld: /tmp/ghdl-dbd129.o: relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile with -fPIC
/tmp/ghdl-dbd129.o: error adding symbols: Bad value
clang: error: linker command failed with exit code 1 (use -v to see invocation)
Makefile:11: recipe for target 'ghdl.so' failed
make: *** [ghdl.so] Error 1
```

tgingold, 2017-02-11:
On 10/02/17 23:05, Juan Gonzalez-Gomez wrote:
> I've tested the new Makefile. It seem that there is still a problem with
> how the libghdlsynth library is compiled
>
> $ make GHDL_PREFIX=/usr/local
> yosys-config --exec --cxx --cxxflags --ldflags -o ghdl.so -shared ghdl/ghdl.cc -DYOSYS_ENABLE_GHDL -I/usr/local/include /usr/local/lib/libghdlsynth.so -Wl,-rpath,/usr/local/lib --ldlibs
> /usr/bin/ld: /tmp/ghdl-dbd129.o: relocation R_X86_64_32 against `.rodata.str1.1' can not be used when making a shared object; recompile
> with -fPIC
> /tmp/ghdl-dbd129.o: error adding symbols: Bad value
> clang: error: linker command failed with exit code 1 (use -v to see
> invocation)
> Makefile:11: recipe for target 'ghdl.so' failed
> make: *** [ghdl.so] Error 1

Ah, maybe it's a new requirement from ubuntu 16.

Can you add -fPIC before -shared ?

Obijuan, 2017-02-11:
It worked!!!!  :-)  Thanks a lot!!!  Now I am ready to test more examples. I will prepare a hello world example for the [icezum Alhambra board ](https://github.com/FPGAwars/icezum/wiki) and Pull-request it to the project

closed 2017-02-11 by Obijuan

tgingold, 2017-02-11:
On 11/02/17 08:23, Juan Gonzalez-Gomez wrote:
> It worked!!!! :-)

Great.  I have updated the Makefile.  Could you try again to check if 
nothing has been broken ?

> Thanks a lot!!!

You're welcome.

 > Now I am ready to test more examples.
> I will prepare a hello world example for the icezum Alhambra board
> <https://github.com/FPGAwars/icezum/wiki> and Pull-request it to the project

Is there a plan to sell this board ?

Tristan.

Obijuan, 2017-02-11:
> Great. I have updated the Makefile. Could you try again to check if nothing has been broken ?

Tested!  Perfect!  ;-)

> Is there a plan to sell this board ? 

Not yet. We are manufacturing a second batch of 250 boards. We will give you one as a gift for your great work. I've been using ghdl for many many years, with my students. Now, thanks to @cliffordwolf and you, we can turn the designs into reality :-)  I really appreciate it a lot

tgingold, 2017-02-11:
On 11/02/17 10:15, Juan Gonzalez-Gomez wrote:
>     Great. I have updated the Makefile. Could you try again to check if
>     nothing has been broken ?
>
> Tested! Perfect! ;-)

Nice.

>     Is there a plan to sell this board ?
>
> Not yet. We are manufacturing a second batch of 250 boards.

Wow.  That's definitely not anymore a prototype build.

> We will give
> you one as a gift for your great work. I've been using ghdl for many
> many years, with my students. Now, thanks to @cliffordwolf
> <https://github.com/cliffordwolf> and you, we can turn the designs into
> reality :-) I really appreciate it a lot

You're welcome.  Nice to have happy users.

Tristan.