ghdl/ghdl-yosys-plugin #4: Error when geting a bits out of a counter

Obijuan, 2017-02-11:
The following vhdl code provokes a segmentation fault when trying to synthesize:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity counter8 is
  port (clk : in std_logic;
        led0 : out std_logic);
end counter8;

architecture synth of counter8 is
  
begin

  process (clk)
    variable temp : unsigned (7 downto 0);
  begin
    if rising_edge(clk) then
      temp:= temp + 1;
      led0 <= temp(0);
    end if;
  end process;

end synth;
```

The command I've used for the synthesis are:

```sh
$ ghdl -a counter8.vhdl
$ yosys -m ../../ghdl.so -p 'ghdl counter8; synth_ice40 -blif counter8.blif'
/----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2016  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7+98 (git sha1 e6cc67b, clang 3.8.0-2ubuntu4 -fPIC -Os)

-- Running command `ghdl counter8; synth_ice40 -blif counter8.blif' --

1. Executing GHDL.
Segmentation fault (core dumped)
```

It takes some seconds to crash

My system is Ubuntu 16.04.1 LTS, running on an 64-bits intel

Obijuan, 2017-02-12:
I have a smaller program that produce the same results:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity vector is
  port (led0: out std_logic);
end vector;

architecture synth of vector is

signal v : std_logic_vector(7 downto 0);

begin
  v <= std_logic_vector'("10101010");
  led0 <= v(0);
end synth;
```

juanmard, 2017-02-12:
Same error results with a simple bit... and another information:

``` vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity no_vector is
  port (led0: out std_logic);
end no_vector;

architecture synth of no_vector is

signal nv : std_logic;

begin
  nv <= '1';
  led0 <= nv;
end synth;
```

Loop in screen with this:

``` sh
Yosys 0.7+111 (git sha1 cdb6ceb8, clang 3.8.1-17 -fPIC -Os)

-- Running command `ghdl vector; synth_ice40 -blif vector.blif' --

1. Executing GHDL.
Importing module vector.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 3) >= this->size() (which is 2)
terminate called recursively
terminate called recursively
terminate called recursively
```

And finally _segmentation fault_

tgingold, 2017-02-12:
This is supposed to work. I will investigate.

Obijuan, 2017-02-12:
Same error with the example provide by @juanmard 

```
ghdl -a vector.vhdl
yosys -m ../../ghdl.so -p 'ghdl vector; synth_ice40 -blif vector.blif'

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2016  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7+98 (git sha1 e6cc67b, clang 3.8.0-2ubuntu4 -fPIC -Os)

-- Running command `ghdl vector; synth_ice40 -blif vector.blif' --

1. Executing GHDL.
Importing module vector.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 3) >= this->size() (which is 2)
terminate called recursively
terminate called recursively
...
Makefile:8: recipe for target 'vector.blif' failed
make: *** [vector.blif] Segmentation fault (core dumped)

```

juanmard, 2017-02-12:
I think i've found it...
I have modified "ghdl.cc" and I have had this

``` sh
Cells and connect.
Instance: 3  Name: \vector%0   Id_Output
Instance: 4  Name: \vector.v     Id_Signal -> sig: 4

terminate called after throwing an instance of 'std::out_of_range'^M
  what():  vector::_M_range_check: __n (which is 4) >= this->size() (which is 3)^M
terminate called recursively
terminate called recursively
....
```
In this line of "ghdl.cc"...

``` C++
log("Id_Signal -> \t");
Net sig = get_driver(get_input(inst, 0));
log (" sig: %d\n", sig.id);
Wire *w = net_map.at(sig.id);
if (w)
   module->rename(w, to_str(iname));
}
```
So I sopouse that _**net_map.at**_ break the code. 
I hope it helps you.

tgingold, 2017-02-13:
I can reproduce the crash.  Investigating.

Obijuan, 2017-02-14:
I can confirm that [commit d343b8fc2861db2dce9d860d517c6cc83acbdc75](https://github.com/tgingold/ghdlsynth-beta/commit/d343b8fc2861db2dce9d860d517c6cc83acbdc75) fixes the problem  reported by @juanmard. This example worked for me:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity no_vector is
  port (led0: out std_logic);
end no_vector;

architecture synth of no_vector is

signal nv : std_logic;

begin
  nv <= '1';
  led0 <= nv;
end synth;
```

The problem with the vector assignment is still present

tgingold, 2017-02-14:
On 14/02/17 05:35, Juan Gonzalez-Gomez wrote:
> I can confirm that commit d343b8fc2861db2dce9d860d517c6cc83acbdc75
> <https://github.com/tgingold/ghdlsynth-beta/commit/d343b8fc2861db2dce9d860d517c6cc83acbdc75>
> fixes the problem reported by @juanmard <https://github.com/juanmard>.
> This example worked for me:
>
> library ieee;
> use ieee.std_logic_1164.all;
> use ieee.numeric_std.all;
>
> entity no_vector is
>   port (led0: out std_logic);
> end no_vector;
>
> architecture synth of no_vector is
>
> signal nv : std_logic;
>
> begin
>   nv <= '1';
>   led0 <= nv;
> end synth;
>
> The problem with the vector assignment is still present

Yes.  I have a fix for it.

I am setting up a testsuite in ghdlsynth to avoid regressions.

Tristan.

tgingold, 2017-02-14:
Fixed with a commit in ghdl.

closed 2017-02-14 by tgingold