ghdl/ghdl-yosys-plugin #10: Test case: Vector8-test1: synth_assignment: cannot handle IIR_KIND_INDEXED_NAME

Obijuan, 2017-02-18:
The following test case throws the error: "synth_assignment: cannot handle IIR_KIND_INDEXED_NAME"

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity vector8_test1 is
  port (led7: out std_logic);
end vector8_test1;

architecture synth of vector8_test1 is

signal v : std_logic_vector(7 downto 0);

begin
  v(7) <= '1';
  led7 <= v(7);
end synth;
```

tgingold, 2017-02-18:
On 18/02/17 06:32, Juan Gonzalez-Gomez wrote:
> The following test case throws the error: "synth_assignment: cannot
> handle IIR_KIND_INDEXED_NAME"
>
> library ieee;
> use ieee.std_logic_1164.all;
> use ieee.numeric_std.all;
>
> entity vector8_test1 is
>   port (led7: out std_logic);
> end vector8_test1;
>
> architecture synth of vector8_test1 is
>
> signal v : std_logic_vector(7 downto 0);
>
> begin
>   v(7) <= '1';
>   led7 <= v(7);
> end synth;

This one is not yet handled.  As long as the index is a constant, it 
shouldn't be the most difficult to implement.

Obijuan, 2017-02-19:
ok. I will continue writing simple test-cases to be used for testing the implementation of the synthesizer
If you don't mind, I will create issues for all the not yet handled messages that I find

I can also help with the implementation, but I should study a little bit more how the current code works

Thanks a lot!  :-)

tgingold, 2017-02-19:
That's ok.  I need issues to understand the priorities.

tgingold, 2019-07-19:
Indexed assignments are now handled.

closed 2019-07-19 by tgingold