ghdl/ghdl-yosys-plugin #11: ERROR: Unsupported instances of logic gates.

C47D, 2017-02-25:
Hi, thanks for the work you are doing. I'm implementing some very basic designs with VHDL for the iCEStick board, here's my code:
``` vhdl
library ieee;
use ieee.std_logic_1164.all;

entity test_or is port (
    sel0, sel1: in std_logic;
    c: out std_logic);
end test_or;

architecture synth of test_or is
begin

    c <= sel1 or sel0;

end synth;
```

My Makefile
```
PROJ = test_or
PIN_DEF = $(PROJ).pcf
DEVICE = hx1k

all: $(PROJ).rpt $(PROJ).bin

%.blif: %.vhdl
	ghdl -a $(PROJ).vhdl
	yosys -q -m $(GHDL)/ghdl.so -p 'ghdl $(PROJ); synth_ice40 -blif $@'

%.asc: $(PIN_DEF) %.blif
	arachne-pnr -q -d $(subst hx,,$(subst lp,,$(DEVICE))) -o $@ -p $^

%.bin: %.asc
	icepack $< $@

%.rpt: %.asc
	icetime -d $(DEVICE) -mtr $@ $<

prog: $(PROJ).bin
	iceprog $<

sudo-prog: $(PROJ).bin
	@echo 'Executing prog as root!!!'
	sudo iceprog $<

clean:
	rm -f $(PROJ).blif $(PROJ).asc $(PROJ).rpt $(PROJ).bin work-obj93.cf

.SECONDARY:
.PHONY: all prog clean
```

The resulting error:
```
ghdl -a test_or.vhdl
yosys -q -m ~/ghdlsynth-beta/ghdl.so -p 'ghdl test_or; synth_ice40 -blif test_or.blif'
ERROR: Unsupported(1): instance \test_or%1 of $or.
Makefile:8: recipe for target 'test_or.blif' failed
make: *** [test_or.blif] Error 1
```

Searching on the ghdl.cc file of this repository seems like the or cell hasn't been implemented yet, the same error happens for xor gate.

Any thoughts?

Thanks in advance
- Carlos

tgingold, 2017-02-26:
Thanks.  Gates or and xor are now supported.

closed 2017-02-26 by tgingold

C47D, 2017-02-26:
Thanks for the quick fix.

tgingold, 2017-02-26:
On 26/02/17 19:41, Carlos Diaz wrote:
> Thanks for the quick fix.

You're welcome.

As you know, this is very WIP.  I plan to add more features, but bugs 
could be fixed quickly.

Tristan.

C47D, 2017-02-27:
Just tested all the basic gates, and, or, xor, nand, nor, not and xnor and all seems to be working properly.

Thanks again
Carlos

commit d85deb6a (2017-02-26): Support or and xor gates

Fix #11