ghdl/ghdl-yosys-plugin #12: Annotate_concurrent_statements_list: cannot handle IIR_KIND_CONCURRENT_SELECTED_SIGNAL_ASSIGNMENT

C47D, 2017-02-27:
Hi,
Did a small design using with-select-when concurrent selective declaration, the code is as follows:
``` vhdl
library ieee;
use ieee.std_logic_1164.all;

entity lut is port (
    sel: in std_logic_vector (1 downto 0);
    c: out std_logic);
end lut;

-- sel(1)  sel(0) |  c 
--   0       0    |  1    
--   0       1    |  0
--   1       0    |  1
--   1       1    |  0

architecture synth of lut is
begin

with sel select c <=

    '1' when "00",
    '0' when "01",
    '1' when "10",
    '0' when others;

end synth;
```

The error i got:
``` bash
$ make
ghdl -a lut.vhdl
yosys -q -m ~/ghdlsynth-beta/ghdl.so -p 'ghdl lut; synth_ice40 -blif lut.blif'
annotate_concurrent_statements_list: cannot handle IIR_KIND_CONCURRENT_SELECTED_SIGNAL_ASSIGNMENT (lut.vhdl:18:1)
```

The full project is attached, on the Makefile GHDL variable points to ghdlsynth-beta directory.

Carlos

[LUT.zip](https://github.com/tgingold/ghdlsynth-beta/files/802531/LUT.zip)

tgingold, 2017-03-01:
Concurrent assignments are not yet handled.  Will add to my TODO list.
(As a workaround, you can rewrite it using a process).

C47D, 2017-03-01:
Hi,
Thanks for the suggestion, i'm still learning, the following code is successfully synthesized and works as expected:
``` vhdl
library ieee;
use ieee.std_logic_1164.all;

entity lut is port (
    sel: in std_logic_vector (1 downto 0);
    c: out std_logic);
end lut;

-- sel(1)  sel(0) |  c 
--   0       0    |  1    
--   0       1    |  0
--   1       0    |  1
--   1       1    |  0

architecture synth of lut is
begin

process (sel) begin
    case sel is
        when "00" => c <= '1';
        when "01" => c <= '0';
        when "10" => c <= '1';
        when "11" => c <= '0';
        when others => c <= '1';
    end case;
end process;

end synth;
```

Does it seems right?
Carlos

tgingold, 2017-03-01:
On 01/03/17 20:00, Carlos Diaz wrote:
> Hi,
> Thanks for the suggestion, i'm still learning, the following code is
> successfully synthesized and works as expected:
>
> library ieee;
> use ieee.std_logic_1164.all;
>
> entity lut is port (
>     sel: in std_logic_vector (1 downto 0);
>     c: out std_logic);
> end lut;
>
> -- sel(1) sel(0) | c
> -- 0 0 | 1
> -- 0 1 | 0
> -- 1 0 | 1
> -- 1 1 | 0
>
> architecture synth of lut is
> begin
>
> process (sel) begin
>     case sel is
>         when "00" => c <= '1';
>         when "01" => c <= '0';
>         when "10" => c <= '1';
>         when "11" => c <= '0';
>         when others => c <= '1';
>     end case;
> end process;
>
> end synth;
>
> Does it seems right?

Yes, this looks ok.

Tristan.

tgingold, 2019-07-21:
Now handled.

closed 2019-07-21 by tgingold