ghdl/ghdl-yosys-plugin #13: "Unsupported" Standard Logic Types

C47D, 2017-03-01:
Doing other little project i find out that some standard types have no support yet, the types that give me the following error 

> ERROR: Unsupported(1): instance \types%1 of $const_UL32.

 are:

- '-' (don't care)
- 'U' (undefined)
- 'X' (unknown strong value)
- 'Z' (high impedance)
- 'H' (weak 1)

I had not seen U, X and H used on the book i am reading but i guess Z and - are used more.

Thanks in advance
Carlos

[test.zip](https://github.com/tgingold/ghdlsynth-beta/files/811772/test.zip)

tgingold, 2017-03-01:
On 01/03/17 20:07, Carlos Diaz wrote:
> Doing other little project i find out that some standard types have no
> support yet, the types that give me the following error
>
>     ERROR: Unsupported(1): instance \types%1 of $const_UL32.
>
> are: [...]

Yes, none of them are currently supported.  Enhancement request.

Tristan.

tgingold, 2019-07-28:
I think this is now fixed by pr #28

C47D, 2019-07-28:
Thanks @tgingold, it has been a while since the issue was opened, i have no access to my fpga at the moment, but i will close this issue and re-open it in case of any trouble. Thanks for your work :)

closed 2019-07-28 by C47D