ghdl/ghdl-yosys-plugin #14: Segfaults with current yosys upon (some) errors.

donnie-j, 2017-08-31:
Hi Tristan,

Thanks for starting down this path.  I was able to compile it against mainline yosys and ghdl today without (problematic) issues on Mac OS X.  I can verify that the included examples synthesize properly with this build.

Unfortunately, the (perhaps not so) simple test case I use for VHDL on ICE40 fails.  Of course, I realize this is just a start at synthesis, but I thought I would open an issue since the constructs used here are the same sorts of things that we use in the J-Core project... which is of course developed with GHDL as the simulator.  I would be great to have a complete open synthesis chain for J-Core projects.

ibuprofen:yosys_example.bundle jeff$ cat top.vhd 
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.top_pack.all;

entity top is port (
   clk :  in std_logic;
   D   : out std_logic_vector(1 to 5));
end top;

architecture beh of top is

signal this_c : top_reg_t;
signal this_r : top_reg_t;
signal rst : std_logic := '0';

begin
   led : process(this_r, clk)
   variable this : top_reg_t;
   variable en : std_logic;
   begin
      this := this_r;

      en := '0';
      if this.prescale < 5000000 then en := '1'; end if;
      this.y := to_slv(this.count, this.blip, en);

      if this.prescale > 5999999 then
         this.prescale := 0;
         this.blip := '1';
         if this.count = 3 then
            this.count := 0;
         else
            this.count := this.count + 1;
         end if;
      else
         if this.prescale = 1000000 then this.blip := '0'; end if;
         this.prescale := this.prescale + 1;
      end if;

      this_c <= this;
   end process;

   led_r0 : process(clk, rst)
   begin
      if rst = '1' then
         this_r <= TOP_REG_RESET;
      elsif clk = '1' and clk'event then
         this_r <= this_c;
      end if;
   end process;

   D <= this_r.y;
end beh;
ibuprofen:yosys_example.bundle jeff$ cat top_pkg.vhd 
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package top_pack is

type top_reg_t is record
  prescale : integer range 0 to (2**24)-1;
  count    : integer range 0 to 3;
  blip     : std_logic;
  y        : std_logic_vector(1 to 5);
end record;

constant TOP_REG_RESET : top_reg_t := ( 0, 0, '0', (others => '0') );

function to_slv(C:integer; B:std_logic; E:std_logic) return std_logic_vector;

component top port (
   clk :  in std_logic;
   D   : out std_logic_vector(1 to 5));
end component;

end package;

package body top_pack is

function to_slv(C:integer; B:std_logic; E:std_logic) return std_logic_vector is
variable ret : std_logic_vector(1 to 5) := (others => '0');
begin
   ret(C+1) := E;
   ret(5)   := B;

   return ret;
end to_slv;

end top_pack;
ibuprofen:yosys_example.bundle jeff$ cat top.vhd 
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.top_pack.all;

entity top is port (
   clk :  in std_logic;
   D   : out std_logic_vector(1 to 5));
end top;

architecture beh of top is

signal this_c : top_reg_t;
signal this_r : top_reg_t;
signal rst : std_logic := '0';

begin
   led : process(this_r, clk)
   variable this : top_reg_t;
   variable en : std_logic;
   begin
      this := this_r;

      en := '0';
      if this.prescale < 5000000 then en := '1'; end if;
      this.y := to_slv(this.count, this.blip, en);

      if this.prescale > 5999999 then
         this.prescale := 0;
         this.blip := '1';
         if this.count = 3 then
            this.count := 0;
         else
            this.count := this.count + 1;
         end if;
      else
         if this.prescale = 1000000 then this.blip := '0'; end if;
         this.prescale := this.prescale + 1;
      end if;

      this_c <= this;
   end process;

   led_r0 : process(clk, rst)
   begin
      if rst = '1' then
         this_r <= TOP_REG_RESET;
      elsif clk = '1' and clk'event then
         this_r <= this_c;
      end if;
   end process;

   D <= this_r.y;
end beh;
ibuprofen:yosys_example.bundle jeff$ ghdl -a top_pkg.vhd
ibuprofen:yosys_example.bundle jeff$ ghdl -a top.vhd
ibuprofen:yosys_example.bundle jeff$ yosys -p "ghdl top"

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2016  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7+     301 (git sha1 c9a4feba, clang 8.1.0 -fPIC)

-- Running command `ghdl top' --

1. Executing GHDL.
Segmentation fault: 11
ibuprofen:yosys_example.bundle jeff$

Instead of printing an error message, upon failure it simply faults.  Simpler code (e.g. y <= (others => '1'); ) does print an appropriate unsupported message.
J.

tgingold, 2017-09-01:
As you can see, the synthesis is very beta (hence the name).
However, your example is nice: it is small, comes from a real design and uses many features.

I have to work again on ghdlsynth, and I will try to handle your design.

Tristan.

tgingold, 2019-09-26:
Now fixed.

closed 2019-09-26 by tgingold