ghdl/ghdl-yosys-plugin #15: error while patching yosys Makefile: patch does not apply

RCoeurjoly, 2018-01-01:
When trying to apply the yosys.diff patch I get the following error.

```
~/yosys$ git apply yosys.diff
error: patch failed: Makefile:49
error: Makefile: patch does not apply
```

The consequence is that when calling yosys with the GHDL option it does not find it.

```
~/ghdlsynth-beta/icestick$

 yosys -p 'ghdl leds; synth_ice40 -blif fixed1.blif'

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2016  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7+428 (git sha1 9804ebe, gcc 4.9.4-2ubuntu1~14.04.1 -fPIC -Os)

-- Running command `ghdl leds; synth_ice40 -blif fixed1.blif' --
ERROR: No such command: ghdl (type 'help' for a command overview)

```
A possible workaround is to specify the absolute path to GHDL:
` yosys -m ../ghdl.so -p 'ghdl leds; synth_ice40 -blif leds.blif'`

Instead of:

`yosys -p 'ghdl leds; synth_ice40 -blif leds.blif'`

Could we integrate those changes into yosys directly?

Thanks.

tgingold, 2018-01-02:
I have to update the patch (and to work again on ghdlsynth).

tgingold, 2020-05-09:
The supported way is now to use the plugin, so I close this issue.

closed 2020-05-09 by tgingold