ghdl/ghdl-yosys-plugin #21: Synthesis hangs without error

marph91, 2019-02-11:
Hi, I was curious about this project, so I tried to follow the instructions on a linux machine (opensuse 15). It worked well so far, but the synthesis hangs at the first step without any error. The command line output is as follows:
```
> yosys -m /home/prog/ghdlsynth-beta/ghdl.so -p 'ghdl leds; synth_ice40 -blif leds.blif'

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2018  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.8+148 (git sha1 e112d2fb, gcc 7.3.1 -fPIC -Os)

-- Running command `ghdl leds; synth_ice40 -blif leds.blif' --

1. Executing GHDL.
```
I know it's beta code, but do you have an idea why it hangs up?

Is there any future work planned on this project?

Best regards,
Marph

tgingold, 2019-02-11:
No, I don't know why it hangs.  That shouldn't be hard to fix.

At some point I have to resume working on it; in particular handling arrays.

Xiretza, 2019-05-08:
Same behaviour here, I used the example in the README and rand it in GDB:

```
$ ghdl -a /usr/share/ghdlsynth/examples/ice40hx8k/leds.vhdl
$ ghdl -a /usr/share/ghdlsynth/examples/ice40hx8k/spin1.vhdl
$ gdb --args yosys -m /usr/lib/ghdlsynth/ghdl.so -p 'ghdl leds; synth_ice40 -blif leds.blif'

(gdb) r
Starting program: /usr/bin/yosys -m /usr/lib/ghdlsynth/ghdl.so -p ghdl\ leds\;\ synth_ice40\ -blif\ leds.blif
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2018  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.8 (git sha1 UNKNOWN, clang 7.0.1 -march=x86-64 -mtune=generic -O2 -fstack-protector-strong -fno-plt -fPIC -Os)

-- Running command `ghdl leds; synth_ice40 -blif leds.blif' --

1. Executing GHDL.

Program received signal SIGABRT, Aborted.
0x00007ffff78d582f in raise () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff78d582f in raise () from /usr/lib/libc.so.6
#1  0x00007ffff78c0672 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff728299d in __gnat_last_chance_handler () at /build/ghdlsynth-beta-git/src/ghdl/./src/grt/grt-cbinding.c:95
#3  0x00007ffff6f9edcd in ada.exceptions.exception_traces.unhandled_exception_terminate () from /usr/lib64/libgnat-8.so
#4  0x00007ffff6f9eb5d in __gnat_unhandled_except_handler () from /usr/lib64/libgnat-8.so
#5  0x00007ffff712307d in __gnat_cleanupunwind_handler () from /usr/lib64/libgnat-8.so
#6  0x00007ffff7a73f5f in _Unwind_ForcedUnwind_Phase2 (exc=0x555555dedaf0, context=0x7fffffffdbb0, frames_p=0x7fffffffdab8) at /build/gcc/src/gcc/libgcc/unwind.inc:170
#7  0x00007ffff7a747d5 in _Unwind_Resume (exc=0x555555dedaf0) at /build/gcc/src/gcc/libgcc/unwind.inc:243
#8  0x000055555567c09a in main ()
```

Seems like something is quite unhappy, calls `abort()`, but `SIGABRT` is immediatly caught again:

```
(gdb) c
Continuing.

Program received signal SIGABRT, Aborted.
0x00007ffff78d582f in raise () from /usr/lib/libc.so.6
(gdb) c
Continuing.

Program received signal SIGABRT, Aborted.
0x00007ffff78d582f in raise () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff78d582f in raise () from /usr/lib/libc.so.6
#1  0x00007ffff78c0672 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff728299d in __gnat_last_chance_handler () at /build/ghdlsynth-beta-git/src/ghdl/./src/grt/grt-cbinding.c:95
#3  0x00007ffff6f9edcd in ada.exceptions.exception_traces.unhandled_exception_terminate () from /usr/lib64/libgnat-8.so
[... 7 lines cut ...]
#11 <signal handler called>
#12 0x00007ffff78d582f in raise () from /usr/lib/libc.so.6
#13 0x00007ffff78c0672 in abort () from /usr/lib/libc.so.6
#14 0x00007ffff728299d in __gnat_last_chance_handler () at /build/ghdlsynth-beta-git/src/ghdl/./src/grt/grt-cbinding.c:95
#15 0x00007ffff6f9edcd in ada.exceptions.exception_traces.unhandled_exception_terminate () from /usr/lib64/libgnat-8.so
#16 0x00007ffff6f9eb5d in __gnat_unhandled_except_handler () from /usr/lib64/libgnat-8.so
#17 0x00007ffff712307d in __gnat_cleanupunwind_handler () from /usr/lib64/libgnat-8.so
#18 0x00007ffff7a73f5f in _Unwind_ForcedUnwind_Phase2 (exc=exc@entry=0x555555e33210, context=context@entry=0x7fffffffcc90, frames_p=frames_p@entry=0x7fffffffcb98) at /build/gcc/src/gcc/libgcc/unwind.inc:170
#19 0x00007ffff7a74550 in _Unwind_ForcedUnwind (exc=0x555555e33210, stop=0x7ffff7123060 <__gnat_cleanupunwind_handler>, stop_argument=<optimized out>) at /build/gcc/src/gcc/libgcc/unwind.inc:217
#20 0x00007ffff6f9eb0d in ada.exceptions.exception_propagation.propagate_gcc_exception () from /usr/lib64/libgnat-8.so
#21 0x00007ffff6f9eb23 in ada.exceptions.exception_propagation.propagate_exception () from /usr/lib64/libgnat-8.so
#22 0x00007ffff6f9fa75 in ada.exceptions.raise_from_signal_handler () from /usr/lib64/libgnat-8.so
#23 <signal handler called>
#24 0x00007ffff78d582f in raise () from /usr/lib/libc.so.6
#25 0x00007ffff78c0672 in abort () from /usr/lib/libc.so.6
[... cut ...]
#29 0x00007ffff712307d in __gnat_cleanupunwind_handler () from /usr/lib64/libgnat-8.so
#30 0x00007ffff7a73f5f in _Unwind_ForcedUnwind_Phase2 (exc=0x555555dedaf0, context=0x7fffffffdbb0, frames_p=0x7fffffffdab8) at /build/gcc/src/gcc/libgcc/unwind.inc:170
#31 0x00007ffff7a747d5 in _Unwind_Resume (exc=0x555555dedaf0) at /build/gcc/src/gcc/libgcc/unwind.inc:243
#32 0x000055555567c09a in main ()
```

This causes the "hanging" behaviour, it's basically just continuously calling signal handlers and slowly filling the stack. Let me know if you need any more info.

marph91, 2019-07-20:
This issue is not anymore reproducible with the current master.

closed 2019-07-20 by marph91

tgingold, 2019-07-20:
Yes, work on the project has resumed (However I haven't yet checked all issues).
Feel free to report any issue, but don't forget it's still work in progress.