ghdl/ghdl-yosys-plugin #22: Unable to install

steinQF, 2019-05-03:
I cannot install this interesting piece of software following the instructions in the README. The problem (besides a typo, and extraneous trailing `d` in `make install.libghdlsynth.included`) is that the patch for `yosys` seems to be compatible with version 0.7 of `yosys` only, which tries to pull `ABC` from an outdated repository which has moved meanwhile.

What is the current plan regarding migration to a more recent version of `yosys`? Is this package still maintained?

Thanks for providing it in the first place!

tgingold, 2019-05-04:
This was just a proof of concept, so do not expect much of it.
The generation of the netlist is still maintained, but the interface with yosys was not.

However, a GSoC project has been proposed so there might be progress during this summer.

tgingold, 2019-06-06:
I haven't updated the patch against yosys, but you can build as a module

1138-4EB, 2019-08-13:
@tgingold, should we close either this or #15 as duplicates?

tgingold, 2019-08-16:
Yes, closing as duplicate.

closed 2019-08-16 by tgingold