ghdl/ghdl-yosys-plugin #24: Initialization of signals and variables doesn't work

marph91, 2019-07-20:
Reproducer: Executing the commands from "spin1" example.

```bash
> ghdl -a leds.vhdl
> ghdl -a spin1.vhdl
> yosys -m /home/prog/ghdlsynth-beta/ghdl.so -p 'ghdl leds; synth_ice40 -json leds.json'
...
-- Running command `ghdl leds; synth_ice40 -json leds.json' --

1. Executing GHDL.
Importing module leds.
ERROR: Unsupported(1): instance \leds.nrst of $isignal.
```

When removing the initializations, the synthesis works.

iamaiy, 2019-07-25:
Not for me

> user@ubuntu: /ghdlsynth-beta/icestick$ ghdl -c
> user@ubuntu: /ghdlsynth-beta/icestick$ ghdl -a leds.vhdl spin1.vhdl
> spin1.vhdl:4:10:warning: declaration of "leds" hides entity "leds" [-Whide]
> user@ubuntu: /ghdlsynth-beta/icestick$ yosys -m ../ghdl.so -p 'ghdl leds'
>  Yosys 0.8+612 (git sha1 be3d9c8e, clang 7.0.0-3 -fPIC -Os)
> -- Running command 'ghdl leds' --
> 1. Executing GHDL.
> Importing module leds.
> ERROR: Unsupported(1): instance \10.cnt of $isignal.

marph91, 2019-07-26:
Did you remove the variable initialization at line 9, too?

`variable cnt : unsigned (1 downto 0);`

tgingold, 2019-07-27:
This is now fixed.  Thanks!

closed 2019-07-27 by tgingold