ghdl/ghdl-yosys-plugin #26: Integer ranges and some integer comparisons don't work

marph91, 2019-07-27:
**MWE:**
```bash
ghdl -a int_test.vhdl
yosys -m /home/prog/ghdlsynth-beta/ghdl.so -p 'ghdl int_test; synth_ice40'
```

`int_test.vhdl`:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity int_test is
  port (clk : in std_logic;
        a, b : in integer range 0 to 1;
        c : out std_logic);
end int_test;

architecture rtl of int_test is
begin
  process (clk)
  begin
    if rising_edge (clk) then
      if a < b then
        c <= '0';
      end if;
    end if;
  end process;
end rtl;
```

**Output:**
```
-- Running command `ghdl int_test; synth_ice40' --

1. Executing GHDL.
ERROR: vhdl import failed.
```

When removing the integer range:
```
-- Running command `ghdl int_test; synth_ice40' --

1. Executing GHDL.
int_test.vhdl:15:12:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_INTEGER_LESS
ERROR: vhdl import failed.
```

When removing the integer range and replacing "less than" with "equality", the synthesis succeeds.

1138-4EB, 2019-08-06:
The same error produced with `icestick/leds.vhdl` and `icestick/spin1.vhdl`, while `icestick/leds.vhdl` and `icestick/blink.vhdl` work ok.

1138-4EB, 2019-08-13:
> The same error produced with `icestick/leds.vhdl` and `icestick/spin1.vhdl`, while `icestick/leds.vhdl` and `icestick/blink.vhdl` work ok.

It seems to be fixed now: https://twitter.com/obijuan_cube/status/1035166655373078530

@marph91 can you please check with the latest master branch from ghdl/ghdl?

marph91, 2019-08-13:
It works for me. Thank you.

However, I didn't test with the current master, because the changes of https://github.com/ghdl/ghdl/commit/7e91b1419cde2a00c9288c02cda256a054ed6a3f seem to be not compatible with this repository yet.

1138-4EB, 2019-08-13:
I saw that after commenting here (see https://github.com/tgingold/ghdlsynth-beta/pull/31#issuecomment-520854715). However, it compiles if you discard the latest two commits of today (https://github.com/ghdl/ghdl/commit/9d61a62f96dc4897dadbf88f5f4ee199d20e0f8f). This is what you did, I assume.

Would you mind closing this? I believe that the mismatch with ghdl/ghdl will be fixed soon.

marph91, 2019-08-14:
Yes, I did take an older commit. I will close, since the issue is resolved.

closed 2019-08-14 by marph91