ghdl/ghdl-yosys-plugin #27: yosys vhdl import fails

iamaiy, 2019-07-27:
I'm trying to run yosys with ghdlsynth on code from my own repo:
https://guthib.com/iamaiy/vhd2icestorm

The code runs with Larry's vhd2vl and the yosys tools.
When i run
user@ubuntu:/vhd2icestorm/vhd$ ghdl -a uart.vhd reset.vhd top.vhd
user@ubuntu:/vhd2icestorm/vhd$ yosys -m ~/ghdlsynth-beta/ghdl.so -p 'ghdl top'

Yosys 0.8+612 (git sha1 be3d9c8e, clang 7.0.0-3 -fPIC -Os)
-- Running command `ghdl top' --
1. Executing GHDL.
ERROR: vhdl import failed.

tgingold, 2019-08-08:
There are some assignments that aren't yet handled.  I am working on it.

trcwm, 2019-09-14:
Hi @tgingold 

How can I help to improve this? I have a design that won't import. Is there a way to get information on which construct it is failing?

tgingold, 2019-09-15:
If your design is large, you can try to synthesize unit per unit (you need to have default to the generics).

If a single unit is failing without a clear message, just open an issue.  It takes me a couple of minutes to find the reason.

trcwm, 2019-09-15:
The design is small - only one VHDL file. Unfortunately, I can't share it so opening an issue with a minimal example isn't possible.

steamka90, 2019-09-21:
Hi @tgingold,

I was trying to run the ghdlsynth with yosys on my DFF example code here:
[https://gitlab.com/steamka90/fusesoc_example](https://gitlab.com/steamka90/fusesoc_example)

However, I'm getting the following:

```
[slee@manjaro-1804-pc vhdl]$ ls
dff.vhd
[slee@manjaro-1804-pc vhdl]$ yosys -m ghdl -p "ghdl -fpsl --std=08 dff.vhd -e dff"
...
 Yosys `0.9+431` (git sha1 779ce353, clang 8.0.1 -fPIC -Os)

-- Running command `ghdl -fpsl --std=08 dff.vhd -e dff' --

1. Executing GHDL.
synth_expression_with_type: cannot handle IIR_KIND_SIGNAL_DECLARATION (dff.vhd:17:10)
ERROR: vhdl import failed.
```
In the DFF source code, I do have the following PSL statements to try to run it with Symbiyosys.
```
    -- set all declarations to run on clk
    default clock is rising_edge(clk);
    d_in_check : assert always {d} |=> {q_int};
    not_d_in_check : assert always {not d} |=> {not q_int};
```

tgingold, 2019-09-21:
Can you post the whole dff.vhd file ?  I think I know how to fix the crash, but I'd like to test before pushing the commits.
Thanks!

steamka90, 2019-09-21:
No problem. :)

```
library ieee;
use ieee.std_logic_1164.all;

entity dff is
  generic(
    formal_g : boolean := true
  );
  port(
    reset : in std_logic;
    clk : in std_logic;
    d : in std_logic;
    q : out std_logic
  );
end entity dff;

architecture rtl of dff is
  signal q_int : std_logic;
begin

  dff_proc : process(clk, reset)
  begin
    if reset = '1' then
      q_int <= '0';
    elsif rising_edge(clk) then
      q_int <= d;
    end if;
  end process dff_proc;

  -- drive q_int to output port
  q <= q_int;

  formal_gen : if formal_g = true generate
  begin
    -- set all declarations to run on clk
    default clock is rising_edge(clk);
    d_in_check : assert always {d} |=> {q_int};
    not_d_in_check : assert always {not d} |=> {not q_int};
  end generate formal_gen;

end rtl;
```

closed 2019-09-21 by tgingold

steamka90, 2019-09-22:
Thank you! I can confirm it works with Yosys now.