ghdl/ghdl-yosys-plugin #36: Synthesizing a single port BRAM

marph91, 2019-08-20:
```vhdl
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity bram is
  generic (
    addr_width : integer := 9;
    data_width : integer := 8
  );
  port (
    clk      : in std_logic;
    we       : in std_logic;
    addr     : in std_logic_vector(addr_width-1 downto 0);
    data_in  : in std_logic_vector(data_width-1 downto 0);
    data_out : out std_logic_vector(data_width-1 downto 0)
  );
end bram;

architecture rtl of bram is
  type mem_type is array (0 to (2**addr_width)-1) of std_logic_vector(data_width-1 downto 0);
  signal mem : mem_type;

begin
  process(clk)
  begin
    if rising_edge(clk) then
      if we = '1' then
        mem(to_integer(unsigned(addr))) <= data_in;
      end if;
    end if;
  end process;
  data_out <= mem(to_integer(unsigned(addr)));
end rtl;
```

Trying to synthesize the code for inferring a BRAM, I found the following three issues:

1. `bram.vhd:20:34:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_INTEGER_EXP`
    - caused by `(2**addr_width)-1`
    - workaround is replacing this by a constant value, i. e. 511
2. `ERROR: Unsupported(1): instance \6 of $dyn_insert.`
    - caused by `to_integer(unsigned(addr))`
    - workaround is replacing this by a constant value
3. `ERROR: vhdl import failed.`
    - caused by `mem_type`
    - workaround is using a std_vector for example

Is this kind of issue okay or should I split it into three independent issues with MWE?

tgingold, 2019-08-20:
It's ok, but do not expect a quick fix.  It's a feature that is planned.

(Thank you for all the reports, that's useful to know the priorities).

marph91, 2019-08-20:
Since it's a leisure project, I don't need a quick fix. Any fix is fine. With the changes so far, it was already possible to synthesize and run simple UART on the icestick :)

1138-4EB, 2019-08-20:
> With the changes so far, it was already possible to synthesize and run simple UART on the icestick :)

That sounds interesting! Would you like to contribute an example to https://github.com/tgingold/ghdlsynth-beta/tree/master/icestick?

marph91, 2019-08-20:
Sure, but I would like to improve the code quality first. There are still some bugs. I will open a PR as soon as they are fixed.

tgingold, 2019-10-01:
This is now handled by `ghdl --synth`.

Xiretza, 2019-10-02:
The only thing that still needs to be done for this to work seems to be handling of `Id_Dyn_Extract` and `Id_Dyn_Insert`. I don't see any gates in `RTLIL::Module` that directly correspond to these, only `concat` and `mux` (`slice` only does constant offsets).

tgingold, 2019-11-07:
Now handled.  Thank you for the issue.

marph91, 2019-11-07:
Thank you :)

I have one more question: When synthesizing the MWE for an `ice40`, the output is

```
> yosys -m ghdl -p 'ghdl bram; synth_ice40'

...

2.42.4. Printing statistics.

=== bram ===

   Number of wires:               4505
   Number of wire bits:           8111
   Number of public wires:         517
   Number of public wire bits:    4123
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               8092
     SB_DFFE                      4104
     SB_LUT4                      3988
```

It seems like the block RAM isn't inferred, but instead LUT are used. Is this the expected behaviour for now?

tgingold, 2019-11-07:
Argh, you're right.  I think this is not expected.
I will investigate.

tgingold, 2019-11-08:
I think the read port must also be synchronous, according to ice40 doc.
(But that may not be fully handled by ghdlsynth).

marph91, 2019-11-08:
I think I tried already two things, but I can verify it only today evening:
* Using almost the same code with the lattice icecube toolchain worked. So I think theoretically async read should be possible.
* Using ghdl+yosys and sync read or forcing bram with the attribute 'ram_style' also didn't succeed.

tgingold, 2019-11-08:
Well, with yosys this simple async read bram doesn't generate a bram:

module test (
        input clk, wen, ren, 
        input [7:0] waddr, raddr,
        input [31:0] wdata,
        output [31:0] rdata
);
        reg [31:0] mem [0:255];
        always @(posedge clk) begin
                if (wen)
                        mem[waddr] <= wdata;
        end
        assign rdata = mem[raddr];
endmodule

marph91, 2019-11-08:
You are right about the async read. I wonder how icecube synthesized it into a bram.

However, I took the verilog example from https://www.reddit.com/r/yosys/comments/5aqzyr/can_i_write_behavioral_verilog_that_infers_ice40/. This infers a bram. Here is analogue VHDL code, which doesn't infer a bram:

```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity bram is
  generic (
    addr_width : integer := 8;
    data_width : integer := 32
  );
  port (
    clk      : in std_logic;
    we       : in std_logic;
    re       : in std_logic;
    waddr    : in std_logic_vector(addr_width-1 downto 0);
    raddr    : in std_logic_vector(addr_width-1 downto 0);
    wdata    : in std_logic_vector(data_width-1 downto 0);
    rdata    : out std_logic_vector(data_width-1 downto 0)
  );
end bram;

architecture rtl of bram is
  type mem_type is array (0 to (2**addr_width)-1) of std_logic_vector(data_width-1 downto 0);
  signal mem : mem_type;
begin
  process(clk)
  begin
    if rising_edge(clk) then
      if we = '1' then
        mem(to_integer(unsigned(waddr))) <= wdata;
      end if;
      if re = '1' then
        rdata <= mem(to_integer(unsigned(raddr)));
      end if;
    end if;
  end process;
end rtl;
```

marzoul, 2019-11-08:
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "
> "name": "View Issue"
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",

The issue is probably about the read-after-write behavior the vhdl code
implements.I bet the ice40 block behaves as more common write-after-
read.This is not particularly explicit in the Lattice user guide for
memories, though.Adrien

Xiretza, 2019-11-26:
Excerpt from `yosys -m ghdl -p 'ghdl bram.vhd -e bram; synth_ice40'`:

```
2.24. Executing MEMORY_BRAM pass (mapping $mem cells to block memories).
Processing bram.\26:
  Properties: ports=2 bits=8192 rports=1 wports=1 dbits=32 abits=8 words=256
  Checking rule #1 for bram type $__ICE40_RAM4K_M0 (variant 1):
    Bram geometry: abits=8 dbits=16 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M0: awaste=0 dwaste=0 bwaste=0 waste=0 efficiency=100
    Rule #1 for bram type $__ICE40_RAM4K_M0 (variant 1) accepted.
    Mapping to bram type $__ICE40_RAM4K_M0 (variant 1):
      Write port #0 is in clock domain \clk.
        Mapped to bram port B1.
      Read port #0 is in clock domain !~async~.
        Bram port A1.1 has incompatible clock type.
        Failed to map read port #0.
    Mapping to bram type $__ICE40_RAM4K_M0 failed.
```

`bram.vhd` is the VHDL posted two comments above. For some reason yosys thinks the read port is asynchronous, even though it's clearly not.

tgingold, 2019-12-06:
Now fixed.

closed 2019-12-06 by tgingold

PPlinux, 2022-01-08:
Dear,

can anyone help me using BRAM 512 word, 16 bit as a ROM (RAM preloaded during FPGA power on) 
using GHDL VHDL -> Yosys -> Lattice ICE40 FPGA please ?

Greetings,

Patrick