ghdl/ghdl-yosys-plugin #38: Handle (non-constant) modulo operation

marph91, 2019-08-25:
Hi, I stumbled upon another errors when trying to synthesizing. I will open a few issues with small reproducer. Maybe it's useful as a list of not yet implemented features.

First one is the modulo operation with non-constant values:
```vhdl
entity modulo_test is
  port (
    a : in integer;
    b : out integer;
    c : out integer
  );
end modulo_test;

architecture rtl of modulo_test is
begin
  b <= a mod 8;
  c <= a rem 8;
end rtl;
```

This yields `../modulo_test.vhdl:14:10:error: non-constant mod not supported`, respectively `../modulo_test.vhdl:15:10:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_INTEGER_REM` when commenting line 14 out.

1138-4EB, 2019-08-25:
Is it possible to synthesise a module operation? I don't mean with GHDL, but with any synthesis tool. I'd say that, even if commercial tools support it, the user should explicitly decide the implementation.

marph91, 2019-08-25:
It should be synthesizable. At least in Xilinx Vivado and Lattice Icecube, I had no problem with it. Quoting the answer comment of your link:

> This is untrue. These operators are synthesisable in some tools when the divisor is a constant; not necessarily a power of 2. You may not always like the result, but that's another matter.

So I guess I always used a constant divisor as in the example code.

tgingold, 2019-08-25:
Shouldn't we support only mod/rem by a power of 2 ?

1138-4EB, 2019-08-25:
> It should be synthesizable. At least in Xilinx Vivado and Lattice Icecube, I had no problem with it. 

Vendors sometimes introduce non-standard solutions to some problems which are common among their users. This is called 'market-driven development' or 'money matters'. Conversely, GHDL is a non-profit project, which is known for being almost pedantic with the standard. The approach to handle deviations which are *de facto* standards is to provide some CLI options for simulation (see https://ghdl.readthedocs.io/en/latest/references/ImplementationOfVHDL.html#using-vendor-libraries).

Hence, when I said *'with any synthesis tool'*, I should have specified that I meant 'get equivalent results with any synthesis tool'. I.e., I'm ok with providing off the shelf solutions for 'complex' math operators such as div/mod, sqrt, etc. However, I think that this should be explicitly enabled through an option.

> Quoting the answer comment of your link:
> ...
> So I guess divisions with constant divisor or divident should be synthesizable.

Precisely, the point is *"You may not always like the result, but that's another matter"*. IMHO, it does matter; it does matter a lot. Hardware description should be as deterministic as possible. If the same operator is synthesized with a different structure (which might affect not only area but also latency), other modules/components in the circuit might need to be adapted. When we describe a FF, mux, BRAM, adder or any other *common* component, we know which the output of synthesis should be, no matter which tool we use. Of course, there are optimization phases that might produce different results, but we know that all of them start with 'the same' base structure. I'm afraid this is not the case with div/mod.

> Shouldn't we support only mod/rem by a power of 2 ?

I think we should consider all the possibilities:

- `const mod const`: this is resolved to a constant.
- `var mod 2`: this is the hardwired.
- `var mod 2**n`: this is hardwires plus an adder.
- `var mod var`: this can have multiple implementations that leverage area/timing. The implementation of each vendor needs to be reverse engineered and, if required, provide an option for each of them.

1138-4EB, 2019-08-27:
@nturley, thanks for the reference. When GHDL is used as a frontend for yosys, the task can be delegated, as you suggest. However, this module shares codebase with GHDL's `ghdl --synth`, which is expected to generate EDIF files. Do you know whether `div`/`mod` operators are allowed in EDIF files?

tgingold, 2019-08-27:
That's true but we have to support both `mod` and `rem`. I suppose yosys only supports `rem` and either we have to handle `mod` by ourself (hence this conversation) or we have to extend yosys.

1138-4EB, 2019-08-27:
> That's true but we have to support both `mod` and `rem`. I suppose yosys only supports `rem` and either we have to handle `mod` by ourself (hence this conversation) or we have to extend yosys.

In the ref provided by @nturley, both `addDiv` and `addMod` are shown, but not `addRem`: https://github.com/YosysHQ/yosys/blob/master/kernel/rtlil.h#L1163-L1164

1138-4EB, 2019-09-30:
Multiplication and division by powers of 2 is now supported: ghdl/ghdl#953

tgingold, 2019-09-30:
In fact, any divisor/multiplicand.  But might be rejected later.

tgingold, 2019-10-01:
Now handled.

closed 2019-10-01 by tgingold