ghdl/ghdl-yosys-plugin #40: VHDL 2008 "when" in sequential statements not supported

antonblanchard, 2019-08-27:
Great to see this making progress again! I tried synthesizing a little Open POWER ISA core we've been working on (microwatt). We already use ghdl for simulation. The first thing to trip up was the following example. I'll convert this to an if statement and see how further I can get.

```
library ieee;
use ieee.std_logic_1164.all;

entity testcase is
        port(
                data_in  : in std_ulogic;
                data_out : out std_ulogic
        );
end entity testcase;

architecture behaviour of testcase is
begin
        comb : process(all)
        begin
                data_out <= '1' when data_in = '0' else '0';
        end process;
end architecture behaviour;
```

1138-4EB, 2019-08-27:
> Sorry, this might be my issue when trying to use the docker container. Checking

Synthesis images (`ghdl/synth:*`) were added very recently and I am still making changes to them and fixing issues. If you find them to be broken, do not hesitate to tell us here or open an issue at [ghdl/docker](https://github.com/ghdl/docker).

antonblanchard, 2019-08-27:
Sorry, this might be my issue when trying to use the docker container. Checking

closed 2019-08-27 by tgingold

tgingold, 2019-08-27:
Now implemented.  Thanks!