ghdl/ghdl-yosys-plugin #41: Set CRON job in Travis CI

1138-4EB, 2019-09-03:
Currently, the CI job is only executed when a new commit is pushed to this repo. However, since this project is tightly coupled to ghdl/ghdl, and because there is no explicit mechanism to bind the version, the status of the project is likely to become broken without anyone noticing. Actually, I think that this is what happened in the last 12 days since the job was last run.

I am cloning this project and running `travis.sh` in a GitHub Actions workflow. Everything worked fine until `yosys -m ghdl -p ghdl ../../ice40hx8k/leds.vhdl ../../ice40hx8k/spin1.vhdl -e leds; synth_ice40 -blif out.blif` started to fail (again) a couple of days ago. See `ext -> synth` in https://github.com/1138-4EB/ghdl-docker/runs/210599897

Therefore, I think it'd be good to set a CRON job in https://travis-ci.com/tgingold/ghdlsynth-beta/settings. It will let us know when an update in ghdl breaks ghdlsynth, even though the latter is untouched.

tgingold, 2019-09-03:
Make sense!

1138-4EB, 2019-09-03:
Great! Unfortunately, I cannot do it on my own, and I cannot see when it's done. This is because this repo is in your userspace, not in the org. Just let me know when you do it and I will close this issue. Or, you can close it yourself.

tgingold, 2019-09-03:
Done.

closed 2019-09-03 by tgingold

1138-4EB, 2019-09-04:
FTR, this is working now. Two CRON jobs were already run: https://travis-ci.com/tgingold/ghdlsynth-beta/builds. It is unfortunate that both are failing, but it allows users to know that they should go back a couple of weeks if they want to test this.