ghdl/ghdl-yosys-plugin #42: Bug in sequential assignment to records in process blocks.

benreynwar, 2019-09-19:
The following vhdl code produces incorrect verilog when run through yosys.
`o_stuff` should not have a dependency on it's value from the previous clock cycle, but the generated verilog code shows this dependency.

```
library ieee;
use ieee.std_logic_1164.all;

package stuff_pkg is
  type t_stuff is
  record
    value1: std_logic;
    value2: std_logic;
  end record;
end package;

library ieee;
use ieee.std_logic_1164.all;
use work.stuff_pkg.all;

entity stuff is
  port (
    clk: in std_logic;
    a: in std_logic;
    o_stuff: out t_stuff
    );
end entity;
 
architecture arch of stuff is
begin
  process(clk)
  begin
    if rising_edge(clk) then
      o_stuff.value1 <= '0';
      o_stuff.value2 <= '0';
      if a = '1' then
        o_stuff.value2 <= '1';
      end if;
    end if;
  end process;
  
end arch;
```
produces
```
/* Generated by Yosys 0.9+406 (git sha1 09ac36da, clang 8.0.0-3 -fPIC -Os) */

module stuff(clk, a, o_stuff);
  wire _0_;
  reg _1_;
  reg _2_;
  input a;
  input clk;
  output [1:0] o_stuff;
  always @(posedge clk)
      _1_ <= 1'h0;
  always @(posedge clk)
      _2_ <= _0_;
  assign _0_ = a ? 1'h1 : o_stuff[1];
  assign o_stuff = { _2_, _1_ };
endmodule
```

benreynwar, 2019-09-19:
Weirdly, this error goes away if we swap the order of `value1` and `value2` in the `t_stuff` definition.

tgingold, 2019-09-19:
I am confused. 
When a = '0', value2 keeps its previous value.  So where is the issue ?

1138-4EB, 2019-09-19:
> When a = '0', value2 keeps its previous value. So where is the issue ?

```
o_stuff.value2 <= '0';
if a = '1' then
  o_stuff.value2 <= '1';
end if;
```

```
if a = '1' then
  o_stuff.value2 <= '1';
else
  o_stuff.value2 <= '0';
end if;
```

```
o_stuff.value2 <= '1' when a='1' else '0';
```

```
o_stuff.value2 <= a;
```

Expect for this last one, the other three examples should produce the same result, and none of them should depend on the previous state, isn't it?

tgingold, 2019-09-19:
Ahhh, you're right!

tgingold, 2019-09-19:
But it should have been recently fixed.  Did you update recently ?

benreynwar, 2019-09-19:
My version of ghdl was 6 days old.  Looks like it's been fixed since then!  Thanks!

closed 2019-09-19 by benreynwar