ghdl/ghdl-yosys-plugin #45: Yosys import fails with: Unsupported(1): instance \17 of $const_bit

tmeissner, 2019-09-22:
I have a design (https://github.com/tmeissner/formal_hw_verification/blob/master/vai_reg/vai_reg.vhd) which generates following error when using Symbiyosys on it:

```
$ sby --yosys "yosys -m ghdl" -f symbiyosys.sby 
SBY 20:54:37 [symbiyosys] Removing direcory 'symbiyosys'.
SBY 20:54:37 [symbiyosys] Copy 'vai_reg.vhd' to 'symbiyosys/src/vai_reg.vhd'.
SBY 20:54:37 [symbiyosys] engine_0: smtbmc z3
SBY 20:54:37 [symbiyosys] base: starting process "cd symbiyosys/src; yosys -m ghdl -ql ../model/design.log ../model/design.ys"
SBY 20:54:37 [symbiyosys] base: ERROR: Unsupported(1): instance \15 of $const_bit.
SBY 20:54:37 [symbiyosys] base: finished (returncode=1)
SBY 20:54:37 [symbiyosys] base: job failed. ERROR.
SBY 20:54:37 [symbiyosys] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:54:37 [symbiyosys] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:54:37 [symbiyosys] DONE (ERROR, rc=16)
```

GHDL with `--synth` option is able to evaluate the code, so I assume, the error is sourced by some  code in the ghdlsynth-beta plugin. 

I found that `const_bit`is defined in the `ghdlsynth.h`file, but there is no code in ghdl.cc which handle the ID for Yosys import. I tried to understand how it is done for `Id_Const_UB32` but I don't understand what the `SigSpec()` function does.

nturley, 2019-09-24:
The Yosys Manual does a really good job of explaining what a SigSpec is.
> The RTLIL::SigSpec data type is used to represent signals. The RTLIL::Cell object contains one RTLIL::SigSpec for each cell port.

This includes constant values, signals or signal subsets, or concatenations of the two.

tmeissner, 2019-09-30:
I was able to extract a MWE:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;

architecture beh of test is

  type t_register is array(0 to 7) of std_logic_vector(7 downto 0);
  signal s_register : t_register;

begin

  -- the problem is the next line
  s_register  <= (others => (others => '0'));

end architecture beh;
```

tmeissner, 2019-09-30:
I'm not sure, but I think, `Const_bit` could be handled in a similar way like `Const_UL32`. I've made these changes to ghdl.cc: https://github.com/tmeissner/ghdlsynth-beta/commit/a73a0387896f1e8d36907091b9348f9f376ed28d

With the additions, Yosys don't generates the error anymore.

@tgingold: Maybe you can have a look, if this could work? If yes, I would prepare a pull request.

tmeissner, 2019-09-30:
I've added handling of Id_Const_Log to my branch: https://github.com/tmeissner/ghdlsynth-beta/commit/96e9d0321f6e76e1fa85715b2c054b98485eb03d

tgingold, 2019-10-01:
But Const_Log can accept width larger than 32.

tmeissner, 2019-10-01:
Oh, of course, you are right. So it's not as simple as I thought. I fear it exceeds my knowledge of GHDL to implement that :(

tgingold, 2019-10-02:
I have added Const_Bit.  I suppose you also need Const_Log ?

tgingold, 2019-10-02:
Const_Log is also handled.

closed 2019-10-02 by tgingold