ghdl/ghdl-yosys-plugin #46: Interoperability with Verilog modules

pepijndevos, 2019-09-24:
Currently a component without an implementation just remains a blackbox, which can possibly be overwritten later by reading verilog modules. But this currently only works one way, and only without generics. To support full two-way interoperability, more work is required.

I talked to Clifford about what it would take to interoperate Verilog modules and VHDL entities in Yosys. So it turns out you need to subclass the `RTLIL::Module` class and override its `derive` method, so that a module with specific parameters/generics gets instantiated. It then returns the name of this module for further use. Conversely, the way to use a Verilog module is to call its derive method with the generics, and use the returned module name.

https://github.com/YosysHQ/yosys/blob/057dae4f78ef604964fb12175f5b263bb386da24/kernel/rtlil.h#L1065

tgingold, 2019-09-25:
Thanks.  I suppose we also need to 'export' all vhdl entities so that yosys know about them.

If someone is a volunteer...

pepijndevos, 2019-09-27:
Also interesting: https://github.com/YosysHQ/yosys/pull/1406

I might volunteer... eventually.

pepijndevos, 2019-11-24:
@tgingold I'm actually trying to do this now, but running into problems because I can't figure out how to do things on the GHDL side.

So step one would be that if an existing module is found, rather than throwing an error, it is `derive`d and connected. To do this I need to get the generic parameters from a `GhdlSynth::Module`, but I can't figure out how to do this. It seems `get_param_uns32` is about something else, because `get_nbr_params` returns zero for my generic component.

The second part would be the reverse mapping. First of all, it seems that for generic modules, GHDL creates parameterized names. The first part would be to _also_ create a "default" instance with a plain name that can be referenced from Verilog. Then a custom `RTLIL::Module` needs to be implemented that lets Yosys `derive` new instances with different parameters.

So I need to be able to do three things I've not figured out yet:
* get the generics from a module.
* get the base name of a generic module.
* create new modules with different generics.

I would appreciate if you could point me to the right API to use for these things.

tgingold, 2019-11-24:
There are no generics in a Module.  So you need to get the generics value before, which is in synth-insts.adb 

Modules are synthesized there by synth_instance.  So if a blackbox is found, you have to check if there is already a known verilog module in yosys, and if so instantiate it.

One non-easy part is to keep modularity: it still has to work (obviously without verilog) without yosys.

For the reverse part, you have to create an instance (this is done in the function Build) but starting from the verilog parameters.

tgingold, 2020-04-01:
It is now possible to interop with verilog module when there are parameters.

closed 2020-04-01 by tgingold

pepijndevos, 2020-04-01:
Great!