ghdl/ghdl-yosys-plugin #47: Handling of various Id_Const_* module IDs missing

tmeissner, 2019-09-30:
There are a bunch of `Const_*` IDs defined in `ghdlsynt_gates.h` which aren't handled in ghdlsynth-beta:

-    Id_Const_SB32
-    Id_Const_UB64
-    Id_Const_UL64
-    Id_Const_X
-    Id_Const_Z
-    Id_Const_0
-    Id_Const_Bit (https://github.com/tgingold/ghdlsynth-beta/issues/45)
-    Id_Const_Log

tmeissner, 2019-10-02:
Handling of Id_Const_Bit (https://github.com/tgingold/ghdlsynth-beta/commit/1af53cad6d6c8b7fd3ad8241ee2537e0a9217276) and Id_Const_Log (https://github.com/tgingold/ghdlsynth-beta/commit/366d22bac43caec422b0628c08b5986b0199e4bf) is now implemented by @tgingold :-)

pepijndevos, 2019-11-24:
I've added `Id_Const_SB32` in #61.

It would actually be good to inventory what is there and what is missing more broadly. Maybe this issue could be renamed and extended to include all of `ghdlsynth_gates.h`?

Here is a script that gets the desired output I think:

```bash
cd ghdl
grep -o 'Id_[a-zA-Z0-9_]*' src/synth/ghdlsynth_gates.h | sort -u > ghdlids.txt
cd ghdlsynth-beta 
grep -o 'Id_[a-zA-Z0-9_]*' src/ghdl.cc | sort -u > ghdlids.txt
cd ..
diff -u  ghdl/ghdlids.txt ghdlsynth-beta/ghdlids.txt
```

Which currently gives

- [ ] Id_Addidx
- [ ] Id_Const_0
- [ ] Id_Const_UB64
- [ ] Id_Const_UL64
- [ ] Id_Design
- [ ] Id_Dyn_Extract
- [ ] Id_Dyn_Insert
- [ ] Id_Dyn_Insert_En
- [ ] Id_Free
- [ ] Id_Memidx
- [ ] Id_Mem_Rd_Sync
- [ ] Id_None
- [ ] Id_Rol
- [ ] Id_Ror
- [ ] Id_Sdiv
- [ ] Id_Srem
- [ ] Id_Udiv
- [ ] Id_Umod
- [ ] Id_User_First

I think it's important to cover all these that can realistically show up in synthesized code before thinking about possibly upstreaming this plugin into Yosys, which I'd very much like to see happen.

tgingold, 2019-11-24:
Those related to memories must not reach ghdl.cc
Some (like Rol, Ror) have to be expanded before reaching ghdl.cc

So the remaining gates are the Id_Const gates (easy to add), and div/rem/mod.
I am not sure that the div/rem/mod of verilog (when it exists) match the one from vhdl...

pepijndevos, 2019-11-24:
It seems `Id_Const_UB64` and  `Id_Const_UL64` are unused.

```ada
   --  Should we keep them ?
   pragma Unreferenced (Id_Const_UB64, Id_Const_UL64);
```

pepijndevos, 2019-11-24:
It appears there is no true remainder operator in Verilog or Yosys. So we can't add that at this time. It might be worth checking with Yosys people if this can be added. I suspect there is not a huge demand.

So with #72 I think we're covering all the IDs that we can/should?

pepijndevos, 2019-11-24:
Filed a Yosys issue: https://github.com/YosysHQ/yosys/issues/1523

closed 2020-04-01 by tgingold