ghdl/ghdl-yosys-plugin #51: Testsuite fails

marph91, 2019-10-03:
The ghdlsynth testsuite seems to fail with  tgingold/ghdlsynth-beta@366d22bac43caec422b0628c08b5986b0199e4bf and ghdl/ghdl@9aa8fe8ff96da383b080ac24081bb71a5d468fb3.

Output:
```
-- Running command `ghdl ../../ice40hx8k/leds.vhdl ../../ice40hx8k/spin1.vhdl -e leds; synth_ice40 -blif out.blif' --

1. Executing GHDL.
Importing module leds.
ERROR: Unsupported(1): instance \22 of $edge.
FAILED!
```

Is this reproducible? Is there anywhere a newer travis ci build of the master branch, than the one from two days ago?

Xiretza, 2019-10-03:
Can't reproduce with 366d22b and ghdl/ghdl@9aa8fe8f (both latest master as of now).

1138-4EB, 2019-10-03:
> Is this reproducible? Is there anywhere a newer travis ci build of the master branch, than the one from two days ago?

There is not. However, you can run `./travis.sh` locally.

marph91, 2019-10-03:
Thank you for the quick answers. Something in my local build was corrupted. `make clean` did help.

closed 2019-10-03 by marph91