ghdl/ghdl-yosys-plugin #53: support for numer_std?

kammoh, 2019-10-04:
Are `signed` and `unsigned` operations currently supported?
Using unsigned addition for examples:

```vhdl
signal x0,x1,x2           : unsigned(13 downto 0);
x0 <= x1 + 1;
```
or
```vhdl
x0 <= x1 + x2;
```
gives:
```
ERROR: Assert `n.id != 0' failed in src/ghdl.cc:58.
```
using `shift_left` (or `shift_right`) gives:
```
ERROR: Unsupported(1): instance \11 of $lsl.
```

`ghdl,ghdlsynth-beta,yosys: master`

tgingold, 2019-10-04:
Most of numeric_std is supported.  Try to update both ghdl and ghdlsynth-beta, and do a build from scratch.

kammoh, 2019-10-04:
I tried, no difference. Everything is on master.

```
GHDL 0.37-dev (v0.36-916-g9aa8fe8f) [Dunoon edition]
 Compiled with GNAT Version: 9.1.0
 llvm code generator
```

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity numeric_test is
	port(
		i : in unsigned(10 downto 0);
		o : out unsigned(10 downto 0)
	);
end entity numeric_test;

architecture RTL of numeric_test is
	
begin
	o <= shift_left(i, 3);

end architecture RTL;
```

```
> yosys -m ghdl -p 'ghdl numeric_test.vhd -e numeric_test'

Yosys 0.9+932 (git sha1 9fef1df3, clang 10.0.1 -fPIC -Os)

-- Running command `ghdl numeric_test.vhd -e numeric_test' --

1. Executing GHDL.
Importing module numeric_test.
ERROR: Unsupported(1): instance \4 of $lsl.
```

tmeissner, 2019-10-04:
I think, this can be fixed by adding handling of `$lsl` &  `$lsr` to ghdlsynth-beta. The suitable functions of the Yosys API seem to be one of these:

```c++
RTLIL::Cell* addShl    (RTLIL::IdString name, RTLIL::SigSpec sig_a, RTLIL::SigSpec sig_b, RTLIL::SigSpec sig_y, bool is_signed = false, const std::string &src = "");
RTLIL::Cell* addShr    (RTLIL::IdString name, RTLIL::SigSpec sig_a, RTLIL::SigSpec sig_b, RTLIL::SigSpec sig_y, bool is_signed = false, const std::string &src = "");
RTLIL::Cell* addSshl   (RTLIL::IdString name, RTLIL::SigSpec sig_a, RTLIL::SigSpec sig_b, RTLIL::SigSpec sig_y, bool is_signed = false, const std::string &src = "");
RTLIL::Cell* addSshr   (RTLIL::IdString name, RTLIL::SigSpec sig_a, RTLIL::SigSpec sig_b, RTLIL::SigSpec sig_y, bool is_signed = false, const std::string &src = "");
RTLIL::Cell* addShift  (RTLIL::IdString name, RTLIL::SigSpec sig_a, RTLIL::SigSpec sig_b, RTLIL::SigSpec sig_y, bool is_signed = false, const std::string &src = "");
RTLIL::Cell* addShiftx (RTLIL::IdString name, RTLIL::SigSpec sig_a, RTLIL::SigSpec sig_b, RTLIL::SigSpec sig_y, bool is_signed = false, const std::string &src = ""); 
```

kammoh, 2019-10-04:
Thanks @tmeissner I figured that out right before checking back here and added unsigned versions `Id_Lsl` and `Id_Lsr`. Would try to make a PR after some testing.

tmeissner, 2019-10-05:
@kammoh Good. I also have created a branch with added `Id_Lsl` and `Id_Ls`. I also added `Id_Asr` for handling `Shift_Right(signed, nat)`. If you have a PR for the first two ones, I would make a separate one for `Id_Asr`. 

I also have a test design which formally verifies the additions with SymbiYosys.
@tgingold: Maybe we should add such tests to the testsuite, as they assure not only that a design is synthesizable, but also have correct behaviour.

tgingold, 2019-10-05:
Yes, feel free to add tests in ghdlsynth-beta.  There is really an hole here.

tgingold, 2019-10-07:
I am closing this issue as shift is implemented.  Please open a new issue if you find another unimplemented operator.

closed 2019-10-07 by tgingold