ghdl/ghdl-yosys-plugin #56: Support for Memidx

antonblanchard, 2019-10-07:
ghdlsynth doesn't understand Memidx yet. A simple example:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test2 is
    port (
        d_in: in std_logic_vector(1 downto 0);
        d_out: out std_logic_vector(1 downto 0)
        );
end entity test2;

architecture rtl of test2 is
    constant c : std_logic_vector (7 downto 0) := "10010000";
begin
    d_out <= c(to_integer(unsigned(d_in))+1 downto to_integer(unsigned(d_in)));
end rtl;
```

```
docker run --rm -t -i -v $(pwd):/src -w /src ghdl/synth:beta bash -c "ghdl -a --std=08 test2.vhdl; yosys -m ghdl -p \"ghdl --std=08 test2\""
```
gives
```
ERROR: Unsupported(1): instance \7 of $memidx.
```

tgingold, 2019-10-07:
Yes, support for memories is till a TODO...

pepijndevos, 2019-10-08:
To summarize my understanding of the situation:

Synthesis flattens multidimensional arrays to a single vector, using `$memidx` to generate an index to a `$dyn_extract` block on the flat vector.
This needs to be done because Yosys apparently does not support 2D arrays.

On the other hand, Yosys needs the array to be turned into an `RTLIL::Memory` and `$memrd`/`$memwr` ports for `$dyn_extract`/`$dyn_insert`. However, these are not flat vectors!

There is also an issue because an array can be used in various ways that are not compatible with memories. So we can't simply turn all 2D arrays into memories because they don't support things like assigning ranges or whole arrays. (although doing so on large arrays would require *a lot* of logic, and is likely a mistake)

My proposal was to not flatten arrays when possible, because Yosys memories have a with and depth. But Tristan argued that not all arrays can become memories, so that it'd be more generic to flatten everything and create a pass that infers memory, and unflattens them.

This pass would have to find `$memidx` cells and find all arrays, make sure they are valid memories, and then transform them.

Random thought: Just set a size threshold for when to flatten an when to make a memory. If you have a large 2D array and use it in a way that's not a valid memory but a giant mux tree, this is probably a mistake.

1138-4EB, 2019-10-08:
> Random thought: Just set a size threshold for when to flatten an when to make a memory. If you have a large 2D array and use it in a way that's not a valid memory but a giant mux tree, this is probably a mistake.

Maybe use/require attributes instead/also? E.g.:

```vhdl
attribute ram_style : string;
attribute ram_style of myram : signal is "distributed";
```

tgingold, 2019-11-03:
This one is now fixed.
The reproducer is slightly strange because ``c`` is longer than what could be addressed, and the stride is 1 while the data size is 2.

closed 2019-11-03 by tgingold