ghdl/ghdl-yosys-plugin #60: Using Entity Instantiation

se-bi, 2019-10-11:
I'm trying to synthesize modules using entity instantiation, which fails with:
```
 Yosys 0.9+431 (git sha1 c372e7ba, gcc 8.3.0-6 -fPIC -Os)

-- Running command `ghdl ../../ice40hx8k/leds.vhdl ../../ice40hx8k/spin1.vhdl leds_wrapper.vhdl leds_wrapper_arch_entity_inst.vhdl -e leds_wrapper; synth_ice40 -blif out.blif' --

1. Executing GHDL.
ERROR: vhdl import failed.
FAILED!
```

I provided a [testcase](https://github.com/se-bi/ghdlsynth-beta/tree/develop/testsuite/issueEntityInst) on my fork.

Different order of the files at the `synth`-command would cause an error when analyzing the file containing the instantiation and points to the specific line with the error.

GHDL itself supports entity instantiation.

Is it an issue of the usage of yosys with ghdlsynth or of ghdlsynth itself?

tgingold, 2019-10-14:
I can reproduce the crash.  Will investigate.

tgingold, 2019-10-15:
Now fixed.  Thanks for the issue.

closed 2019-10-15 by tgingold