ghdl/ghdl-yosys-plugin #65: Comparing port with unassigned signal gives error

Munken, 2019-10-30:
If an input port is compared with a signal, which is not utilized within an entity architecture,
then it fails with:

```
ERROR: Assert `n.id != 0' failed in src/ghdl.cc:67.
```

Example with std_logic_vector:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity latch is
  port (
    signal clk          : in std_logic;
    signal strobe       : in std_logic;
    signal data         : in std_logic_vector(7 downto 0)
    );
end entity;

architecture rtl of latch is
  signal other         : std_logic_vector(7 downto 0) := (others => '0');        
begin
  
  default clock is rising_edge(clk);
  assert always {strobe = '1'}
    |=> next (data = other
    until strobe = '0');  
end architecture;
```

Example with std_logic:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity latch is
  port (
    signal clk          : in std_logic;
    signal strobe       : in std_logic;
    signal data         : in std_logic
    );
end entity;

architecture rtl of latch is
  signal other         : std_logic := '0';        
begin
  
  default clock is rising_edge(clk);
  assert always {strobe = '1'}
    |=> next (data = other
    until strobe = '0');  
end architecture;
```

SymbiYosys file:
```
[options]
mode bmc
depth 20

[engines]
smtbmc

[script]
ghdl -fpsl --std=08 -gformal=true latch.vhd -e latch
prep -top latch

[files]
latch.vhd
```

Munken, 2019-10-30:
Even simpler example of the error:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity latch is
  port (
    signal clk          : in std_logic;
    signal data         : in std_logic
    );
end entity;

architecture rtl of latch is
  signal other         : std_logic := '0';        
begin
  
  default clock is rising_edge(clk);
  assert always {true}
    |=> next (data = other);  
end architecture;
```

tgingold, 2019-10-30:
Yes, initialized signals that are never assigned are not yet handled.

tgingold, 2019-11-07:
Now fixed.  Thank you for the issue.

closed 2019-11-07 by tgingold