ghdl/ghdl-yosys-plugin #68: demux inferred as memory

donnie-j, 2019-11-04:
Tristan, thanks again for GHDL and working to make opensource synthesis possible!

I've been testing multiple pieces of code, trying to get a feel for what is still to be done before we can synthesize [j-core for ICE40](https://github.com/j-core/j-core-ice40) with GHDL+Yosys.

Here is a fairly self contained reproducer, for a problem with inference of a demux:

`library ieee;
use ieee.std_logic_1164.all;

entity demux is port (
   j :  in integer range 0 to 4;
   k :  in std_logic;
   l :  in std_logic;
   y : out std_logic_vector(1 to 5));
end demux;

architecture beh of demux is

   function to_slv(C:integer; B:std_logic; E:std_logic) return std_logic_vector is
   variable ret : std_logic_vector(1 to 5) := (others => '0');
   begin
      ret(C+1) := E;
      ret(5)   := B;

      return ret;
   end to_slv;
begin
   y <= to_slv(j, k, l);
end beh;
`
This seems to result inference of memory for y, which is of course not the intent.  This works with other ICE40 tools, including verific+yosys (last I checked).

`jeff@JeffNB:~/work/tools/blink$ ghdl -a t.vhd 
jeff@JeffNB:~/work/tools/blink$ ghdl -e demux
jeff@JeffNB:~/work/tools/blink$ yosys -m ghdl

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+932 (git sha1 81876a37, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)

yosys> ghdl demux
1. Executing GHDL.
t.vhd:18:7:note: dynamic write to a non-memory
Importing module demux.
ERROR: Unsupported(1): instance \16 of $memidx.
ERROR: Assert `GetSize(ports) >= it.second->port_id' failed in kernel/rtlil.cc:1353.
jeff@JeffNB:~/work/tools/blink$`

Thanks again... I'll try and make smallish test cases for other issues found also, if that is helpful.

tgingold, 2019-11-04:
I think this is now fixed after a commit in ghdl.  Will watch before closing this issue.

Thanks!

tgingold, 2019-11-04:
Test is ok!

closed 2019-11-04 by tgingold