ghdl/ghdl-yosys-plugin #69: Bounds check with complex types?

donnie-j, 2019-11-04:
Sorry for not boiling this down further.  To be more clear, because this is not a small reproducer, I don't expect this will magically all work once past the reported assert...

This bug seems to be triggered by a few different parts of the j-core code.

spi_pkg.vhd
```library ieee;
use ieee.std_logic_1164.all;

package spi_phy_pack is

type spi_phy_dir_t is (IDLE, MO, MI);

type spi_phy_i_t is record
   dir  : spi_phy_dir_t;
   d    : std_logic_vector(7 downto 0);
   en   : std_logic;
end record;

type spi_phy_o_t is record
   d    : std_logic_vector(7 downto 0);
   ack  : std_logic;
   v    : std_logic;
   dir  : spi_phy_dir_t;
end record;

end package;
```

xip_ctl_pkg.vhd
```library ieee;
use ieee.std_logic_1164.all;
use work.spi_phy_pack.all;

package xip_pack is

constant XIP_ADDR_BITS : natural := 24;
constant XIP_LINE_BITS : natural := 10;
constant XIP_ROWS_BITS : natural := 2;

constant XIP_RAM_BITS  : natural := XIP_LINE_BITS + XIP_ROWS_BITS;
constant XIP_TAG_BITS  : natural := XIP_ADDR_BITS - XIP_RAM_BITS;
constant XIP_ROWS      : natural := 2**XIP_ROWS_BITS;

type xip_tag_t is record
   v    : std_logic;
   tag  : std_logic_vector(XIP_TAG_BITS-1 downto 0);
end record;
constant XIP_TAG_RESET : xip_tag_t := ( '0', (others => '0') );

type xip_tag_ram_t is array (0 to XIP_ROWS-1) of xip_tag_t;

type xip_i_t is record
   en  : std_logic;
   a   : std_logic_vector(XIP_ADDR_BITS-1 downto 2);
end record;

type xip_o_t is record
   ack : std_logic;
   d   : std_logic_vector(31 downto 0);
end record;

type xip_state_t is ( IDLE, MISS, CMD, ADDR, DUMMY, FILL );

type xip_reg_t is record
   ram : xip_tag_ram_t;
   s   : xip_state_t;
   cnt : integer range 0 to (2**XIP_LINE_BITS)-1;
   y   : xip_o_t;
   ym  : spi_phy_i_t;
end record;

constant XIP_REG_RESET : xip_reg_t := ( (others => XIP_TAG_RESET), IDLE, 0,
                                        ( '0', (others => '0') ),
                                        ( IDLE, (others => '0'), '0') );

type xip_ram_t is array (0 to 2**(XIP_LINE_BITS+XIP_ROWS_BITS-2)-1) of std_logic_vector(31 downto 0);

component xip_ctl is port (
   clk : in  std_logic;
   rst : in  std_logic;
   a   : in  xip_i_t;
   y   : out xip_o_t;
   am  : in  spi_phy_o_t;
   ym  : out spi_phy_i_t);
end component;

end package;
```

xip_ctl.vhd
```library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use work.spi_phy_pack.all;
use work.xip_pack.all;

entity xip_ctl is port (
   clk : in  std_logic;
   rst : in  std_logic;
   a   : in  xip_i_t;
   y   : out xip_o_t;
   am  : in  spi_phy_o_t;
   ym  : out spi_phy_i_t);
end xip_ctl;

architecture beh of xip_ctl is

signal this_c : xip_reg_t;
signal this_r : xip_reg_t;
signal read : std_logic_vector(31 downto 0);

signal ram    : xip_ram_t;

begin

   p1 : process(a, this_r, am)
   variable this : xip_reg_t;
   variable tag  : std_logic_vector(XIP_TAG_BITS -1 downto 0);
   variable row  : integer range 0 to (2**XIP_ROWS_BITS)-1;
   variable ra   : std_logic_vector(23 downto 0);
   begin
      this := this_r;
      tag  := a.a(XIP_ADDR_BITS-1 downto XIP_LINE_BITS+XIP_ROWS_BITS);
      row  := to_integer(unsigned(a.a(XIP_LINE_BITS+XIP_ROWS_BITS-1 downto XIP_LINE_BITS)));
      ra   := (others => '0');
      ra  (XIP_ADDR_BITS-1 downto XIP_LINE_BITS) := a.a(XIP_ADDR_BITS-1 downto XIP_LINE_BITS);

      this.y.ack := '0';
      -- check for hit 
      if a.en = '1' and this.s = IDLE then
         if this.ram(row).v = '1' and this.ram(row).tag = tag then
            this.y.ack        := '1';
         else
            this.ram(row).v   := '0';
            this.ram(row).tag := tag;
            this.s            := CMD;
         end if;
      end if;

      if this.s = CMD and am.dir = IDLE then
         this.ym.dir  := MO;
         this.ym.d    := x"0b";
         this.ym.en   := '1';
         this.s  := ADDR;
         this.cnt:= 0;
      elsif this.s = ADDR and am.ack = '1' then
         if    this.cnt = 0 then this.ym.d    := ra  (23 downto 16);
         elsif this.cnt = 1 then this.ym.d    := ra  (15 downto  8);
         elsif this.cnt = 2 then this.ym.d    := ra  ( 7 downto  0);
         else  this.s   := DUMMY; this.ym.dir  := MI;
         end if;
         this.cnt := this.cnt + 1;
      elsif this.s = DUMMY and am.v = '1' then
         this.s       := FILL;
         this.cnt     := 0;
      elsif this.s = FILL and am.v = '1' then
         if this.cnt = (2**XIP_LINE_BITS)-1 then
            this.s    := IDLE;
            this.ram(row).v := '1';
            this.ym.en:= '0';
         else
            this.cnt  := this.cnt + 1;
         end if;
      end if;

      this_c <= this;
   end process;

   p2 : process(rst, clk)
   begin
      if rst = '1' then
         this_r <= XIP_REG_RESET;
      elsif clk'event and clk = '0' then
         this_r <= this_c;
      end if;
   end process;

   r  : process(clk, a, am, this_r)
   variable waddr : integer range 0 to 2**(XIP_LINE_BITS+XIP_ROWS_BITS-2)-1;
   variable raddr : integer range 0 to 2**(XIP_LINE_BITS+XIP_ROWS_BITS-2)-1;
   variable byte  : integer range 0 to 3;
   begin
      raddr := to_integer(unsigned(a.a(XIP_LINE_BITS+XIP_ROWS_BITS-1 downto 2)));
      if clk'event and clk = '0' then
         read <= ram(raddr);
      end if;

      waddr := to_integer(unsigned(a.a(XIP_LINE_BITS+XIP_ROWS_BITS-1 downto XIP_LINE_BITS)) & 
                          to_unsigned(this_r.cnt/4, XIP_LINE_BITS-2));
      byte  := this_r.cnt mod 4;
      if clk'event and clk = '0' and am.v = '1' and this_r.s = FILL then
         if byte = 0 then ram(waddr)(31 downto 24) <= am.d; end if;
         if byte = 1 then ram(waddr)(23 downto 16) <= am.d; end if;
         if byte = 2 then ram(waddr)(15 downto  8) <= am.d; end if;
         if byte = 3 then ram(waddr)( 7 downto  0) <= am.d; end if;
      end if;
   end process;

   -- outputs
   ym <= this_r.ym;
   y  <= (ack => this_r.y.ack, d => read);
end beh;
```

Synthesis results
```jeff@JeffNB:~/work/tools/tests$ ghdl -a spi_pkg.vhd xip_ctl_pkg.vhd xip_ctl.vhd 
jeff@JeffNB:~/work/tools/tests$ ghdl -e xip_ctl
jeff@JeffNB:~/work/tools/tests$ yosys -m ghdl

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+932 (git sha1 81876a37, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)

yosys> ghdl xip_ctl
1. Executing GHDL.

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1141-g61d4cdf2) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/home/jeff/work/tools/tests/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-builders.adb:1401
******************************************************************
ERROR: vhdl import failed.

yosys>
```

tgingold, 2019-11-04:
I can reproduce the issue.  This is also due to memories.

tgingold, 2020-01-01:
Now fixed.
Thank you for the issue.

closed 2020-01-01 by tgingold

donnie-j, 2020-01-03:
Wow, that unblocked a lot of things for the J-Core codebase... We're 
going to need to retest and give you a status update :)

We've got a hardware project that works on ICE40 up5k using Lattice 
tools ready to announce as soon as we get (most of the way) there with 
GHDL synth.

Really, congrats.
J.

tgingold, 2020-01-03:
Handling RAMs took more time than expected...