ghdl/ghdl-yosys-plugin #71: yosys module loading error,

donnie-j, 2019-11-15:
It seems there is a mismatch somewhere in the build process (at least as I've been doing it) that cropped up a few days ago (here, abbreviated):

Yosys:
```
make config-gcc
PREFIX=/home/jeff/work/tools/usr make
PREFIX=/home/jeff/work/tools/usr make install
```
GHDL:
```
./configure --enable-libghdl --enable-synth --prefix=/home/jeff/work/tools/usr
make -j4
make install
```
GHDL Synth:
```
export PATH=/home/jeff/work/tools/usr/bin:$PATH
make; make install
```
But something has missing symbols...

```
jeff@JeffNB:~/work/tools/test/spi_fault$ 
jeff@JeffNB:~/work/tools/test/spi_fault$ which ghdl
/home/jeff/work/tools/usr/bin/ghdl
jeff@JeffNB:~/work/tools/test/spi_fault$ ghdl -a spi_pkg.vhd spi.vhd 
jeff@JeffNB:~/work/tools/test/spi_fault$ ghdl -e spi_phy
jeff@JeffNB:~/work/tools/test/spi_fault$ which yosys
/home/jeff/work/tools/usr/bin/yosys
jeff@JeffNB:~/work/tools/test/spi_fault$ yosys -m ghdl

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+932 (git sha1 e907ee4f, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)

ERROR: Can't load module `./ghdl': /home/jeff/work/tools/usr/lib/libghdl-0_37_dev.so: undefined symbol: __ghdl_flag_string
jeff@JeffNB:~/work/tools/test/spi_fault$
```

tgingold, 2019-11-15:
Oops, I broke it recently while adding support for files.  I am fixing that.

tgingold, 2019-11-15:
Should be fixed.

closed 2019-11-15 by tgingold