ghdl/ghdl-yosys-plugin #73: yosys script fails on second ghdl command

Fatsie, 2019-11-29:
A yosys script can have only one ghdl command.
I have the following script `test.ys`:
```
ghdl --std=08 --ieee=synopsys T80se
ghdl --std=08 --ieee=synopsys c4m_jtag_tap_controller
```
I get the following output:
```console
(nmigen) [verhaegs@localhost source]$ yosys -m ghdl -s test.ys

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+932 (git sha1 ff8529a, gcc 4.8.5 -fPIC -Os)

-- Executing script file `test.ys' --

1. Executing GHDL.
/home/verhaegs/eda/Chips4Makers/Retro-uC/asic/C4M035/SnowWhiteIII_Z80/source/t80/T80_Reg.vhd:88:20:note: found RAM "regsh", width: 8 bits, depth: 8
/home/verhaegs/eda/Chips4Makers/Retro-uC/asic/C4M035/SnowWhiteIII_Z80/source/t80/T80_Reg.vhd:89:20:note: found RAM "regsl", width: 8 bits, depth: 8
Importing module t80se.
Importing module t80.
Importing module t80_mcode.
Importing module t80_alu.
Importing module t80_reg.

2. Executing GHDL.

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1310-g8d3c6df) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-pc-linux-gnu
/home/verhaegs/eda/Chips4Makers/Retro-uC/asic/C4M035/SnowWhiteIII_Z80/source/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : libraries.adb:638
******************************************************************
ERROR: vhdl import failed.
```
If I change the order of the ghdl statements in `test2.py`:
```
ghdl --std=08 --ieee=synopsys c4m_jtag_tap_controller
ghdl --std=08 --ieee=synopsys T80se
```
I get the following output:
```console
(nmigen) [verhaegs@localhost source]$ yosys -m ghdl -s test2.ys

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+932 (git sha1 ff8529a, gcc 4.8.5 -fPIC -Os)

-- Executing script file `test2.ys' --

1. Executing GHDL.
Importing module c4m_jtag_tap_controller.
Importing module c4m_jtag_tap_fsm.
Importing module c4m_jtag_irblock.
Importing module c4m_jtag_idblock.
Importing module c4m_jtag_ioblock.
Importing module c4m_jtag_iocell.

2. Executing GHDL.

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1310-g8d3c6df) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-pc-linux-gnu
/home/verhaegs/eda/Chips4Makers/Retro-uC/asic/C4M035/SnowWhiteIII_Z80/source/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : psl-nodes.adb:307
******************************************************************
ERROR: vhdl import failed.
```

pepijndevos, 2019-12-01:
Can you provide a minimal working example? IIRC I've definitely ran multiple GHDL commands in the past.

Fatsie, 2019-12-01:
Here is a minimal **non**-working example. vhdl code:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity cell1 is
  port (
    O: out std_logic
  );
end entity cell1;

architecture rtl of cell1 is
begin
  O <= '0';
end architecture rtl;

library ieee;
use ieee.std_logic_1164.all;

entity cell2 is
  port (
    O: out std_logic
  );
end entity cell2;

architecture rtl of cell2 is
begin
  O <= '1';
end architecture rtl;
```
And the shell script:
```shell
#!/bin/sh
opts='--std=08'

ghdl -a $opts multi_ghdl.vhdl
yosys -m ghdl <<EOF
ghdl $opts cell1
ghdl $opts cell2
EOF
```
Giving output:
```console
[verhaegs@localhost test_ghdl]$ ./multi_ghdl.sh 

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+932 (git sha1 ff8529a, gcc 4.8.5 -fPIC -Os)

yosys> ghdl --std=08 cell1
1. Executing GHDL.
Importing module cell1.

yosys> ghdl --std=08 cell2

2. Executing GHDL.

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1322-g3bd9a18) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-pc-linux-gnu
/tmp/test_ghdl/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : libraries.adb:638
******************************************************************
ERROR: vhdl import failed.

yosys> exit

End of script. Logfile hash: ebc39c5cb9
CPU: user 0.07s system 0.04s, MEM: 153.51 MB total, 8.56 MB resident
Yosys 0.9+932 (git sha1 ff8529a, gcc 4.8.5 -fPIC -Os)
Time spent: 100% 2x ghdl (0 sec)
```
ghdl is compiled with gcc 8.3.0

1138-4EB, 2019-12-01:
Are you trying to synthesise two top level entities at the same time?

Fatsie, 2019-12-01:
Yes, if you have a top design in verilog that instantiates different blocks written in VHDL you need to import each instantiated block separately AFAICS.

Fatsie, 2019-12-01:
In my case top level is a nMigen generated block and the design contains a Z80 and MOS6502 core both written in VHDL together with my VHDL JTAG code.

1138-4EB, 2019-12-01:
I see... but I don't know if that feature is currently supported. I guess that a possible workaround is to wrap all your VHDL blocks in a single entity.

Fatsie, 2019-12-01:
The VHDL is wrapped in a nMigen class and code should be able to freely combine this wrapped VHDL code. Constructing then a single top level for all the VHDL is not straight forward.
I was actually doing this single top entity workaround but hope the feature will be considered for ghdlsynth.

1138-4EB, 2019-12-01:
I'm quite sure that the feature will be implemented eventually. The workaround is just a temporal solution, should you need/want to have it done now.

Fatsie, 2019-12-03:
@tgingold 
I compiled new ghdl version and ghdlsynth-beta but minimal example still fails, now at another point:
```console
[verhaegs@localhost test_ghdl]$ ./multi_ghdl.sh 

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+932 (git sha1 ff8529a, gcc 4.8.5 -fPIC -Os)

yosys> ghdl --std=08 cell1
1. Executing GHDL.
Importing module cell1.

yosys> ghdl --std=08 cell2

2. Executing GHDL.

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1336-gf342fd4) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-pc-linux-gnu
/tmp/test_ghdl/
Command line:

Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-context.adb:332 access check failed
******************************************************************
ERROR: vhdl import failed.

yosys> exit

End of script. Logfile hash: ebc39c5cb9
CPU: user 0.06s system 0.01s, MEM: 153.64 MB total, 8.57 MB resident
Yosys 0.9+932 (git sha1 ff8529a, gcc 4.8.5 -fPIC -Os)
Time spent: 100% 2x ghdl (0 sec)
```

tgingold, 2019-12-04:
Ah ok.  I can reproduce it but only when using --std=08.  It works ok without it.

tgingold, 2019-12-06:
Fixed.

commit 87544a7a (2019-12-03): Support multiple synthesis.  Fix #73
closed 2019-12-06 by tgingold