ghdl/ghdl-yosys-plugin #76: Assert failure in RTLIL::SigSpec::as_const() for T65

Fatsie, 2019-12-03:
I have problems importing T65 from the [chips4makers t65 repo](https://gitlab.com/chips4makers/t65.git).
Unfortunately I wasn't able to make a minimal testbench for the bug but could do some debugging.
I modified ghdlsynth to show in which instance it fails and run the following script:
```sh
ghdl -a --std=08 --ieee=synopsys T65_Pack.vhd T65_ALU.vhd T65_MCode.vhd T65.vhd
ghdl -r --std=08 --ieee=synopsys t65
yosys -m ghdl <<EOF
ghdl --std=08 --ieee=synopsys t65
EOF
```
Some excerpt from the output:
```console
[verhaegs@localhost vhdl]$ ./run.sh 
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/synopsys/std_logic_arith.vhdl:255:20:@0ms:(assertion warning): There is an 'U'|'X'|'W'|'Z'|'-' in an arithmetic operand, the result will be 'X'(es).

...

 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+932 (git sha1 ff8529a, gcc 4.8.5 -fPIC -Os)

yosys> ghdl --std=08 --ieee=synopsys t65
1. Executing GHDL.
../../src/synopsys/std_logic_arith.vhdl:219:30:warning: unknown pragma "synthesis_off" ignored [-Wpragma]
../../src/synopsys/std_logic_arith.vhdl:223:29:warning: unknown pragma "synthesis_on" ignored [-Wpragma]
../../src/synopsys/std_logic_arith.vhdl:225:30:warning: unknown pragma "synthesis_off" ignored [-Wpragma]
../../src/synopsys/std_logic_arith.vhdl:229:29:warning: unknown pragma "synthesis_on" ignored [-Wpragma]
../../src/synopsys/std_logic_arith.vhdl:236:33:warning: unknown pragma "synthesis_off" ignored [-Wpragma]

...

Instance: \498
Instance: \499
Instance: \500
Instance: \501
Instance: \502
Id_(A|Ia)dff
ERROR: Assert `is_fully_const() && GetSize(chunks_) <= 1' failed in kernel/rtlil.cc:3741.
```
This thus seems to be cause by instance 502 which is a iadff. The corresponding output of ``ghdl --synth`` for instance 502 is :
```vhdl
  -- /home/verhaegs/eda/Chips4Makers/Retro-uC/asic/C4M035/SnowWhiteIII_MOS6502/source/t65/T65.vhd:416:16
  inst_502 : gsynth.gate_iadff port map (
    clk => wrap_clk,
    d => n501_o,
    rst => n323_o,
    rst_val => n324_o,
    init => n14_o,
    q => n502_q);
```
I verified and n502_q is connected to P output of t65. The ``ghdl.iadff`` is generated from the source code in _T65.vhd_:
```vhdl
	process (Res_n_i, Clk)
		variable tmpP:std_logic_vector(7 downto 0);--Lets try to handle loading P at mcycle=0 and set/clk flags at same cycle
	begin
		if Res_n_i = '0' then
			P <= x"00"; -- ensure we have nothing set on reset
		elsif Clk'event and Clk = '1' then
			tmpP:=P;
			if (Enable = '1') then
				if (really_rdy = '1') then
					if MCycle = "000" then
						if LDA = '1' then
							ABC(7 downto 0) <= ALU_Q;
						end if;
						if LDX = '1' then
							X(7 downto 0) <= ALU_Q;
						end if;
						if LDY = '1' then
							Y(7 downto 0) <= ALU_Q;
						end if;
						if (LDA or LDX or LDY) = '1' then
							tmpP:=P_Out;
						end if;
					end if;
					if SaveP = '1' then
						tmpP:=P_Out;
					end if;
					if LDP = '1' then
						tmpP:=ALU_Q;
					end if;
					if IR(4 downto 0) = "11000" then
						case IR(7 downto 5) is
						when "000" =>--0x18(clc)
							tmpP(Flag_C) := '0';
						when "001" =>--0x38(sec)
							tmpP(Flag_C) := '1';
						when "010" =>--0x58(cli)
							tmpP(Flag_I) := '0';
						when "011" =>--0x78(sei)
							tmpP(Flag_I) := '1';
						when "101" =>--0xb8(clv)
							tmpP(Flag_V) := '0';
						when "110" =>--0xd8(cld)
							tmpP(Flag_D) := '0';
						when "111" =>--0xf8(sed)
							tmpP(Flag_D) := '1';
						when others =>
						end case;
					end if;
					tmpP(Flag_B) := '1';
					if IR = "00000000" and MCycle = "100" and RstCycle = '0' then
						--This should happen after P has been pushed to stack
						tmpP(Flag_I) := '1';
					end if;
					if SO_n_o = '1' and SO_n = '0' then
						tmpP(Flag_V) := '1';
					end if;
					if RstCycle = '1' then
						tmpP(Flag_I) := '0';
						tmpP(Flag_D) := '0';
					end if;
					tmpP(Flag_1) := '1';

					P<=tmpP;--new way

					SO_n_o <= SO_n;
					if IR(4 downto 0)/="10000" or Jump/="01" then -- delay interrupts during branches (checked with Lorenz test and real 6510), not best way yet, though - but works...
						IRQ_n_o <= IRQ_n;
					end if;
				end if;
				-- detect nmi even if not rdy
				if IR(4 downto 0)/="10000" or Jump/="01" then -- delay interrupts during branches (checked with Lorenz test and real 6510) not best way yet, though - but works...
					NMI_n_o <= NMI_n;
				end if;
			end if;
	    end if;
	end process;
```
I tried to reduce the code but I ended always without ``ghdl.iadff`` in the ``--synth`` output but with using ``rising_edge()`` and also no failure when importing into yosys.

tgingold, 2019-12-03:
The specific issue for iadff is fixed, but it still fails.

The problem is for signals like Y: their value is kept during reset.  I will try to fix that.

tgingold, 2019-12-31:
Now fixed.  I was able to synthesize t65 in yosys.

closed 2019-12-31 by tgingold