ghdl/ghdl-yosys-plugin #80: Assignment of std_logic from generic to part of std_logic_vector unhandled.

thasti, 2020-01-19:
While trying to synthesize a larger design, I stumbled across a limitation in ghdlsynth. Let me preface this by saying that ghdlsynth is already very useful for me and works like a charm for a lot of code I wrote in the past without modification - a bit Thank You for that!

### Description
The bug happens when supplying an initialization value in the form of a std_logic through a generic, which is then later assigned to part of a std_logic_vector. It triggers an `unhandled IIR_PREDEFINED_ELEMENT_ARRAY_CONCAT` in `synth_static_dyadic_predefined` while running yosys ghdl. Assigning the same generic std_logic to another std_logic works, so I guess it is partially fine. This happens on ghdl + ghdlsynth-beta master.

### Testcase
```
library ieee;
use ieee.std_logic_1164.all;

entity testcase is
    generic (
        init_bit : std_logic := '1'
    );
end testcase;

architecture rtl of testcase is
    -- assigning generic to std_logic works OK
    signal test_assign : std_logic := init_bit; 
    -- assigning generic to part of std_logic_vector breaks ghdlsynth
    signal test_assign_vector : std_logic_vector(1 downto 0) := init_bit & "0";
begin
end rtl;
```

### Outcome
```
[thasti@pc tmp]$ ghdl -a testcase.vhd 
[thasti@pc tmp]$ yosys -m ghdl.so
yosys> ghdl testcase
1. Executing GHDL.
testcase.vhd:14:74:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_ELEMENT_ARRAY_CONCAT

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1510-g501193da) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-pc-linux-gnu
/tmp/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-static_oper.adb:434
******************************************************************
ERROR: vhdl import failed.
```

tgingold, 2020-01-20:
Now fixed.  Thank you for the issue.

closed 2020-01-20 by tgingold

thasti, 2020-01-20:
Thanks a lot for looking at this and fixing it! I believe in the process of testing the fix I came across another corner case with the same type of arrangement that is not yet handled gracefully:

```
entity testcase is
    generic (
        init_bit : std_logic := '1'
    );
end testcase;

architecture rtl of testcase is
    -- assigning generic to multiple parts of std_logic_vector breaks ghdlsynth
    signal test_assign_vector : std_logic_vector(2 downto 0) := init_bit & "0" & init_bit;
begin
end rtl;
```
The difference here is that `init_bit` is used twice in the vector concatenation. With this, I'm getting the same error as before:

```
1. Executing GHDL.
testcase.vhd:14:80:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_ARRAY_ELEMENT_CONCAT

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1525-gcaed711f) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-pc-linux-gnu
/tmp/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-static_oper.adb:454
******************************************************************
ERROR: vhdl import failed.
```

thasti, 2020-01-22:
I confirm this is now fixed for my original use case :) Thanks a lot!

eine, 2020-01-22:
@thasti, do you mean that `:= init_bit & "0" & init_bit` works? Or the previous one does but this one is still failing? If so, I'd suggest to either reopen this issue or open a new one.

thasti, 2020-01-22:
Both cases now work, Tristan pushed another fix to ghdl yesterday after my second report.

eine, 2020-01-22:
Thanks for clarifying. It's sometimes hard to follow him ;)