ghdl/ghdl-yosys-plugin #82: Getting TYPES.INTERNAL_ERROR : synth-static_oper.adb:524 in specific files

suoto, 2020-02-09:
Using GHDL Synth to compile https://github.com/phase4ground/dvb_fpga/tree/master/rtl/bch_generated/bch_128x64.vhd fails with the following message

```
root@0a883929b46b:/src# ghdl --synth rtl/bch_generated/bch_128x16.vhd -e bch_128x16
rtl/bch_generated/bch_128x16.vhd:202:61:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_1164_SCALAR_XOR

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1576-g76e2f6b3) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/src/
Command line:
ghdl --synth rtl/bch_generated/bch_128x16.vhd -e bch_128x16
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-static_oper.adb:524
Call stack traceback locations:
0x55e00d157790 0x55e00d14b861 0x55e00d161d10 0x55e00d14b772 0x55e00d161d10 0x55e00d14b772 0x55e00d161d10 0x55e00d14b772 0x55e00d161d10 0x55e00d14b772 0x55e00d161d10 0x55e00d14b772 0x55e00d161d10 0x55e00d14b772 0x55e00d161d10 0x55e00d14b772 0x55e00d161d10 0x55e00d13cc82 0x55e00d13ffb7 0x55e00d148047 0x55e00d140516 0x55e00d14802a 0x55e00d140516 0x55e00d14802a 0x55e00d140516 0x55e00d14057b 0x55e00d14802a 0x55e00d1487cb 0x55e00d14a18d 0x55e00d14a347 0x55e00d13c2ba 0x55e00d13c3d8 0x55e00d135e1d 0x55e00d192eff 0x55e00d0db916 0x55e00d23da9b 0x55e00cea5949 0x7fe0d5691099 0x55e00cea41a8 0xfffffffffffffffe
******************************************************************

```

For the record, all files inside https://github.com/phase4ground/dvb_fpga/tree/master/rtl/bch_generated/ fail in pretty much the same way.

I'm using the ghdl/synth:beta container, versions are

```
root@0a883929b46b:/src# ghdl  --version
GHDL 0.37-dev (v0.36-1576-g76e2f6b3) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
root@0a883929b46b:/src# yosys --version
Yosys 0.9+1706 (git sha1 UNKNOWN, clang 7.0.1-8 -fPIC -Os)
```

Please let me know if you need more info!

eine, 2020-02-09:
@suoto, this repo is about the module that allows to use GHDL as a frontend for yosys. However, it seems that you are testing synthesis features with `ghdl --synth`. That feature is provided by the core ghdl, without requiring yosys. Hence, I would suggest to open/move these issues in ghdl/ghdl instead.

suoto, 2020-02-09:
Yes, sorry about that, will close both issues and reopen on ghdl/ghdl repo. :+1:

By the way, the idea is to use Yosys to synthesize as well, that's why I'm using `ghdl/synth:beta'

closed 2020-02-09 by suoto