ghdl/ghdl-yosys-plugin #84: fatal error: 'map' file not found in src/ghdl.cc

beja65536, 2020-02-11:
Hi everybody, I just cloned the repository and tried to run `make` but ran into the following error:

> yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I/usr/local/include -O
In file included from src/ghdl.cc:19:
/usr/local/share/yosys/include/kernel/yosys.h:42:10: fatal error: 'map' file not found
#include <map>
         ^~~~~
1 error generated.
Makefile:25: recipe for target 'ghdl.o' failed
make: *** [ghdl.o] Error 1

Any ideas what could cause make to fail?
I have installed Yosys according to the readme and its `make test` works fine. GHDL has been installed with the GCC backend and so far I did not had any issues.

Thanks!

Xiretza, 2020-02-11:
What's the output of `yosys-config --exec echo --cxx --cxxflags`? Sounds to me like it's trying to parse the C++ header as C for whatever reason.

beja65536, 2020-02-11:
clang -Wall -Wextra -ggdb -I/usr/local/share/yosys/include -MD -D_YOSYS_ -fPIC -I/usr/local/include -std=c++11 -Os -DYOSYS_ENABLE_READLINE -DYOSYS_ENABLE_PLUGINS -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ZLIB -I/usr/include/tcl8.6 -DYOSYS_ENABLE_TCL -DYOSYS_ENABLE_ABC -DYOSYS_ENABLE_COVER

beja65536, 2020-02-11:
I am currently rebuilding yosys with with config-gcc. I'll let you know if that solves the problem.

Xiretza, 2020-02-11:
For what it's worth, I can't reproduce the problem with those `--cxxflags`. Since `#include <map>` is part of any standard C++ installation, something must be seriously wrong with the clang/C++ stdlib installation though.

beja65536, 2020-02-11:
Rebuilding yosys with the `make config-gcc` solves this problem. If there is time I will investigate in using clang.

closed 2020-02-11 by beja65536