ghdl/ghdl-yosys-plugin #86: Recoding FSM under special condition results in non-functional synthesis result

hackfin, 2020-02-15:
Description: When an FSM in a specific implementation using a state with no initialization value is used, it is recognized by yosys to be recoded as 'one-hot'. Synthesis then produces a non-functional result (for yet unknown reason)

Reproduce:
* Change the lines in `examples/ecp5_versa/fifobuf.vhdl` (look for `GHDLSYNTH_QUIRK`):
```
	-- GHDLSYNTH_QUIRK
	-- Needs this initialized, otherwise gets 'optimized away'
	-- signal state : state_t := S_IDLE;
	-- If we don't initialize, yosys feels like it wants to recode.
	signal state : state_t;
```
* Check `versa_ecp5_top-report.txt` (look for `FSM_RECODE` pass), you might diff against the report of the original commit (#85 27b14ac)
* UART loopback of example no longer working due to dysfunctional state machine

Maybe someone is faster spotting the error. It is very likely related to yosys and not the GHDL front end, however I thought it's worth investigating the entire chain.
I have tried isolating the case by creating a BLIF and converting this back to Verilog for a roundtrip simulation (see attached fifotest.mk). When synthesizing against this Verilog module, the synthesis produces a *working* result.
[fifotest.mk.txt](https://github.com/tgingold/ghdlsynth-beta/files/4208334/fifotest.mk.txt)

ZirconiumX, 2020-02-26:
After importing this into Yosys with `ghdl` can you then run `write_ilang fsm_bug.il` and upload it? It's possible there's a bug in the FSM code that GHDL finds by generating different RTLIL compared to Yosys' internal Verilog parser.

hackfin, 2020-02-27:
Here are the ilang outputs: (working: *_init.il, failing: *_noinit.il)

Currently synthing example from this branch: https://github.com/hackfin/ghdlsynth-beta/tree/masocist

[fifobuf_ilang.zip](https://github.com/tgingold/ghdlsynth-beta/files/4260367/fifobuf_ilang.zip)

tgingold, 2020-03-01:
I had a look at this issue.  Do you have a unit test for fifobuf ?

hackfin, 2020-03-01:
Unfortunately, the unit tests I have so far are for virtual setups only. I'll attempt to run the synthesized result through iverilog and see if the problem exhibits again. I'm in process to set up a co-simulation environment with MyHDL to make it easier to test various implementations. More later.

hackfin, 2020-07-13:
This issue seems to have resolved magically along the way with more recent yosys/ghdl combinations, so I think it's time to close it.
For completeness, the above test setup is found in a recently set up examples playground (launcher to fifobuf test for ghdl plugin: [image])

closed 2020-07-13 by hackfin