ghdl/ghdl-yosys-plugin #87: Bad elaboration order

rroohhh, 2020-02-15:
Hi, 
I am trying to use this with a bigger [project](https://github.com/apertus-open-source-cinema/axiom-beta-firmware/tree/master/peripherals/soc_main) targeting the Xilinx Zynq 7020 FPGA. 

I am using this script: https://hastebin.com/pojonogoca.sh to try and synthesize the design. It produces the following output: https://hastebin.com/raw/ezacoqabek

(The `unisim.vhd` and `unimacro.vhd` come from a `Vivado` installation, they are `Vivado/2018.2/data/vhdl/src/unisim/unisim_retarget_VCOMP.vhd` and `Vivado/2018.2/data/vhdl/src/unimacro/unimacro/unimacro_VCOMP.vhd`)

It fails at the `yosys` step with

```
error: synth: bad elaboration order of objects

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1603-g47da51fc) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/src/
Command line:
 
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-context.adb:221
******************************************************************
ERROR: vhdl import failed.
```

This is of course not a lot of information to work with, but due to the size of the project, I am not sure how do debug / reduce this error further.

Do you have any pointers for me on how to debug this further or reduce the project to something workable?

tgingold, 2020-02-16:
I suppose this is due to the use of generic packages.  This vhdl-08 feature is not yet supported by synth.

cajt, 2020-02-26:
I get the identical error message for a different design. 
It uses VHDL-93, so no generic packages.  
But it has similar (maybe the same) warnings about unknown pragmas from "src/synopsys/std_logic_arith.vhdl".

Shall I try to build a minimal example, and open an new issue?

tgingold, 2020-02-27:
Yes, if you have a vhdl-93 reproducer, please post it. No need to create a new issue.
It doesn't need to be minimal, but it is much better when it is stand-alone.

The warnings in std_logic_arith could be ignored.

cajt, 2020-02-28:
Here is the source: https://github.com/cajt/ghdlsynth-testcase-loa 
and its output: https://raw.githubusercontent.com/cajt/ghdlsynth-testcase-loa/master/setup1/log/output.log

tgingold, 2020-02-28:
I can reproduce the crash.

tgingold, 2020-02-29:
The bad elaboration error is now fixed.

But there are some unimplemented operations that I have to fix.

tgingold, 2020-02-29:
ghdl now detects a latch in uart_rx.

cajt, 2020-02-29:
Thank you. 
I'm trying to reproduce the detected latch, but have to solve some build-environment problems first.
Also I'm moderately sure the design/code was synthesized before and ran on spartan3/6 and ice40 (with Icecube2) some years ago, but I can't reproduce this ad-hoc.

tgingold, 2020-02-29:
A latch is synthesisable, but currently rejected by ghdlsynth.

tgingold, 2020-03-01:
I am closing this issue because the bad elaboration order crash is now fixed.
Do not hesitate to open a new issue if you still have a problem with the design.

closed 2020-03-01 by tgingold

rroohhh, 2020-03-09:
For my case I still get the same `bad elaboration order of objects` error. I updated my initial message with the updated log and script I used. Here they are again:
Script: https://paste.niemo.de/ubehahuwup.sh
Output: https://paste.niemo.de/raw/lazolamaxo

tgingold, 2020-03-09:
The elaboration order crash is now fixed.  There is a different crash after that; I will have a look at it.

rroohhh, 2020-03-21:
Great news, after going through things affected by #93 it seems to atleast not crash or error out. There are still some issues with some cells not being correctly techmapped by yosys, that I will look into further soon.