ghdl/ghdl-yosys-plugin #89: Multiplier inference optimizations

hackfin, 2020-03-09:
When inferring a multiplier, inputs seem to become sign extended to the result's signal width, therefore yosys seems to create (unnecessary) cascades. This is specifically tested for ECP5 multipliers, but should be technology invariant.

Test case added as patch, the expected result is generated by synth_techmap_verilog()

[patch-multiplier.txt](https://github.com/tgingold/ghdlsynth-beta/files/4306286/patch-multiplier.txt)

tgingold, 2020-03-09:
You should perform the 'wreduce' pass before mapping.  

But maybe ghdlsynth shouldn't try to resize operands of multipliers.

tgingold, 2020-04-07:
I am closing this issue as a solution is provided.

closed 2020-04-07 by tgingold