ghdl/ghdl-yosys-plugin #92: GHDL crashes while trying to synthesize ZPU

dkopotev, 2020-03-11:
Hi,
I am currently working on the following project:
```
https://github.com/dkopotev/zpu-lattice
```
Simulation of the code with the standard ghdl works perfectly:
```
mkdir sim/rtl_sim/run
cd sim/rtl_sim/run
../bin/dmips_med1_tb.ghdl.sh
```
But the same code cannot be synthesized with ghdlsynth-beta:
```
mkdir syn/yosys/run
cd syn/yosys/run
../bin/zpu_top.yosys.sh
```
The tool reports no errors. It just crashes:
```
../../../rtl/vhdl/zpu_medium.vhdl:181:29:warning: declaration of "state" hides signal "state" [-Whide]
../../../rtl/vhdl/zpu_medium.vhdl:201:29:warning: declaration of "state" hides signal "state" [-Whide]
../../../rtl/vhdl/zpu_medium.vhdl:228:33:warning: declaration of "state" hides signal "state" [-Whide]

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+1706 (git sha1 UNKNOWN, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)

-- Running command `ghdl zpu_top; synth_ice40 -blif hardware.blif' --

1. Executing GHDL.
../../../rtl/vhdl/txt_util.vhdl:138:27:warning: unknown pragma "translate" ignored [-Wpragma]
../../../rtl/vhdl/txt_util.vhdl:141:27:warning: unknown pragma "translate" ignored [-Wpragma]
../../../rtl/vhdl/phi_io.vhdl:197:29:warning: unknown pragma "translate" ignored [-Wpragma]
../../../rtl/vhdl/phi_io.vhdl:200:29:warning: unknown pragma "translate" ignored [-Wpragma]
../../../rtl/vhdl/phi_io.vhdl:204:33:warning: unknown pragma "translate" ignored [-Wpragma]
../../../rtl/vhdl/phi_io.vhdl:226:33:warning: unknown pragma "translate" ignored [-Wpragma]
../../../rtl/vhdl/bram.vhdl:69:11:note: found RAM "ram", width: 32 bits, depth: 4096

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (tarball) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/home/dima/projects/fpga/lattice/zpu-lattice/syn/yosys/run/
Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-context.adb:531
******************************************************************
ERROR: vhdl import failed.
```
Any idea where the problem is?
Thank you in advance.
Best regards,
Dmitry

Xiretza, 2020-03-11:
This particular issue is caused by [this line](https://github.com/dkopotev/zpu-lattice/blob/6d9c90f7d0c9ecf68ce2b0dcad75da9a8d2f2066/rtl/vhdl/phi_io.vhdl#L198) (found by simply synthesizing one module at a time, then deleting lines in a binary search to find the exact problem). Since there are at least 3 other constructs in there that cause crashes, it's not trivial to just fix everything though. I recommend posting the project over in ghdl/ghdl#974 so it can be tracked and used as a bugfixing target.

dkopotev, 2020-03-11:
Thank you, Xiretza! Now I get it. The tool does not understand `--synopsys translate off` pragma and tries to synthesize unsinthesizable code. I will rework the code.
With regard to the ghdlsynth-beta, the comment could be more meaningful.
By the way, what do you mean by
> deleting lines in a binary search to find the exact problem

tgingold, 2020-03-12:
Yes, '--synopsys translate_off' is understood, but '--synopsys translate off' is not.
I will add it.

Xiretza, 2020-03-12:
@dkopotev I just deleted half the lines, checked if it still crashed the same way, and repeated.

tgingold, 2020-03-22:
It doesn't crash anymore if ENA_LOG generic of ZPUPhiIO is set to false.

closed 2020-04-07 by tgingold