ghdl/ghdl-yosys-plugin #93: error: clocked logic requires clocked logic on else part

rroohhh, 2020-03-20:
I am still trying to get the project from #87 working. With the latest updates I am getting an error I actually managed to minimize. When trying to synthesize this file:
```vhdl
library IEEE;
use IEEE.std_logic_1164.ALL;

entity A is
  port (
    clk : in std_logic;
        input : in std_logic;
        output : out std_logic
    );

end entity A;

architecture RTL of A is
begin

    not_proc : process (clk)

        variable not_input : std_logic := '0';

    begin
        if rising_edge(clk) then
      not_input := not input;
    end if;

    output <= not_input;
    end process;
end RTL;
```
using `yosys -m ghdl -p "ghdl -frelaxed-rules --std=08 --ieee=synopsys t.vhd -e A"`

It fails with 
```
 Yosys 0.9+1706 (git sha1 UNKNOWN, clang 7.0.1-8 -fPIC -Os)

-- Running command `ghdl -frelaxed-rules --std=08 --ieee=synopsys t.vhd -e A' --

1. Executing GHDL.
t.vhd:22:8:error: clocked logic requires clocked logic on else part
ERROR: vhdl import failed.
```

This is using `GHDL release: 1.0-dev (v0.37.0-105-gf7991678) [Dunoon edition]`

Is this actually invalid vhdl? Vivado seems to accept it.

eine, 2020-03-20:
Shouldn't `output <= not_input` be either outside of the process or inside the if/else clause?

rroohhh, 2020-03-20:
Yes both of these make the example be accepted by ghdl in the case. Do you know if the two versions you proposed are actually completely equivalent to my example?

tgingold, 2020-03-21:
I agree the example should pass.

eine, 2020-03-21:
> Yes the first two of these make the example be accepted by ghdl in the case. Do you know if the three versions you proposed are actually completely equivalent to my example? (I guess atleast the third one should be, but that one actually fails in ghdl with the same error message)

I tend to avoid variables for synthesis, so I'm not confident in this area. Anyway, I believe that all three examples are equivalent, and yours too. I would have written the following, which should also be equivalent:

```vhdl
architecture RTL of A is
begin
  not_proc : process (clk)
  begin
    if rising_edge(clk) then
      output <= not input;
    end if;
  end process;
end RTL;
```

rroohhh, 2020-03-22:
Well this is a simplified example so the specific behaviour of a variable (especially that assignments update the value of a variable immediately) would not be as simple to be folded into the logic.

closed 2020-03-22 by tgingold