ghdl/ghdl-yosys-plugin #95: The link to the Docker image in README.md point to no where

rodrigomelo9, 2020-03-26:
Hello, my name is Rodrigo and I am starting with ghdlsynth (but I used ghdl for simulation really a lot, since 2006).

I following the link to the Docker Image and I reach a blank page. Is the idea point to the tags list? I found it at https://hub.docker.com/r/ghdl/synth/tags. Also, I think that has no cost to add the command to pull the image, for people with no docker knowledge. Something like:

```
Docker image [`ghdl/synth:beta`](https://hub.docker.com/r/ghdl/synth/tags) (`docker image pull ghdl/synth:beta`) includes yosys...
```

I can make the changes and generate a pull request, but I was not sure exactly where to point :P

eine, 2020-03-26:
Hola Rodrigo!

> I am starting with ghdlsynth (but I used ghdl for simulation really a lot, since 2006).

That will hopefully make it easy for you to pick synthesis features, since all the parsing and analysis is common to simulation and synthesis. I.e., you are already aware of which language features are supported and which are not yet.

> I following the link to the Docker Image and I reach a blank page. Is the idea point to the tags list? I found it at [hub.docker.com/r/ghdl/synth/tags](https://hub.docker.com/r/ghdl/synth/tags).

Yes, [hub.docker.com/r/ghdl/synth/tags](https://hub.docker.com/r/ghdl/synth/tags) is the correct URL today. However, it is quite annoying that docker (the company) keeps "breaking the internet" by changing/modifying the URLs... At first, we used `hub.docker.com`. Then, when huge investors stepped in and they tried to make money from a SaaS or PaaS, it was changed to `cloud.docker.com`, and `hub` stopped working. It seems that, since the company was split some months ago, `hub.docker.com` is the canonical URL again.

Hence, a PR to fix it today is welcome! Just bear in mind that it might be broken again in some months.

> Also, I think that has no cost to add the command to pull the image, for people with no docker knowledge. Something like:
> 
> ```
> Docker image [`ghdl/synth:beta`](https://hub.docker.com/r/ghdl/synth/tags) (`docker image pull ghdl/synth:beta`) includes yosys...
> ```

I don't think this is necessary:

- The image is pulled automatically when used, in case it is not available locally already. 
- The pull command is shown in hub.docker.com/r/ghdl/synth/tags.
- Even if users have very little knowledge about docker, they need to know the essence: you download/pull images, you use/execute/run containers.
- Pulling ghdl/synth:beta only is almost useless. P&R, bitstream generation and programming do require other sibling images.
- @antonblanchard wrote a Makefile for ECP5 targets which uses either native installations, docker or podman: [antonblanchard/ghdl-yosys-blink](https://github.com/antonblanchard/ghdl-yosys-blink). See also ghdl/docker#21. @hackfin used it as a base to contribute the `*.mk` files in subdir [examples](https://github.com/tgingold/ghdlsynth-beta/tree/master/examples) of this repo.

Furthermore, this repository is expected to be either partially upstreamed to yosys, moved to organisation ghdl (say ghdl/yosys-plugin) and/or merged as a subdir of ghdl/ghdl. Likely, all of them at the same time. See #74. In this context:

- All the docker-related content is gathered in [ghdl/docker](https://github.com/ghdl/docker).
- ghdlsynth-beta (renamed to yosys-plugin or whatever) will be documented in https://ghdl.readthedocs.io/en/latest/development/Synthesis.html.
- I would like to extend https://github.com/ghdl/docker/blob/master/USE_CASES.md to include screencasts of synthesis and P&R examples. Then, move it to ghdl.rtfd.io. See also ghdl/docker#15.

Last, but not least, I believe that all the `ghdl/cache` and `ghdl/synth` images that are created in workflow [cache](https://github.com/ghdl/docker#-cache-5-jobs--max-4--11-images-weekly) from ghdl/docker, should be upstreamed elsewhere. I suggested it to maintainers of YosysHQ, but it didn't gather any attention.

Fortunately, @mithro (from [SymbiFlow](https://symbiflow.github.io)) is open to creating a `docker` repo in organisation [github.com/SymbiFlow](https://github.com/SymbiFlow/). I think that the first step to boost such a collaboration/migration is to modify the diagram in [SymbiFlow project structure](https://symbiflow.github.io/#about), so that ghdl and/or ghdlsynth-beta are shown. Then, it should be considered whether explanations related to docker/podman/OCI for new users should better fit in the context of symbiflow, rather than in ghdl.rtfd.io. In the end, GHDL is one tool in a stack of 3-4 steps that are required to either configure/program a device or do formal verification.

> I can make the changes and generate a pull request, but I was not sure exactly where to point :P

I think it is ok to fix the URL. It is also ok to add any further explanation that helps new users get familiar with the makefiles in subdir `examples`. Just bear in mind that the information available in the README of this repo is insufficient on purpose. We don't want users (those who are not familiar with the structure of the organisation) to take this as a primary source of info, because it will likely be gone in some weeks/months.

rodrigomelo9, 2020-03-26:
Wow, a big explanation, thanks.

Ok, I will change the URL and be aware of future movements.

I was [comparing Yosys against ISE and Vivado](https://github.com/rodrigomelo9/yosys-versus), using examples provided by Xilinx. I am now adding the analysis of these files (the VHDL version) with ghdlsynt :-)

eine, 2020-03-26:
> I was [comparing Yosys against ISE and Vivado](https://github.com/rodrigomelo9/yosys-versus), using examples provided by Xilinx. I am now adding the analysis of these files (the VHDL version) with ghdlsynt :-)

Note that it is not necessarily Yosys vs ISE/Vivado. It is possible to use Yosys and Vivado; GHDL and ISE/Vivado; or GHDL and Yosys and ISE/Vivado. This is because ISE/Vivado include multiple tools in a single suite. Synthesis is just one of the steps, which can be replaced by external tools such as Yosys or Mentor's Precision. Hence, you can use open source tools for parsing/analysis/synthesis, but keep the vendor tools for P&R and bitstream generation.

Overall, I think that what you are trying to do is interesting. It is similar to [ymherklotz/verismith](https://github.com/ymherklotz/verismith), where they try to find synthesis errors/bugs in multiple tools. However, although IANAL, I don't know if you are allowed to use those sources for that purpose...

rodrigomelo9, 2020-03-27:
What I was doing was Yosys vs ISE/Vivado :P I didn't consider the possibility of another synthesis tool (today I noticed about GHDL synthesis features, beyond ghdlsynth).

The project that you mentioned is GPL3 for non-comercial projects, so I can use it... but if they already checking against Yosys... Anyway, thanks for the data ;-)

mithro, 2020-03-27:
@eine - BTW You might be interested in taking a look at https://j.mp/openfpga-diagram and [Status of SymbiFlow Flows Google Doc](https://docs.google.com/document/d/1Q8vaqD67aqPO7JbSDZtVvbg4BD7XBVuA_JJTwcYHprk/edit).

eine, 2020-03-27:
> The project that you mentioned is GPL3 for non-comercial projects, so I can use it... but if they already checking against Yosys... Anyway, thanks for the data ;-)

@rodrigomelo9, I meant the fact that you are using HDL sources from Xilinx to test a different tool. Some vendors distribute HDL sources with a license that allows you to use them with their tools only, and to target their devices only. Some licenses are also locked to a particular board/device. Hence, you might not be allowed to synthesise Xilinx's HDL sources with Yosys. Honestly, I find it pointless, but they are entitled to do so.

Regarding the comparison with verismith, IIRC they are trying to find corner cases where Verilog is ambiguous, in order to trigger hard-to-debug issues in the synthesis tools. Hence, I believe that your idea is still useful. Precisely, inference of BRAMs is being complex for Tristan. He started to work on it six months ago, and it is still an issue.

I thought that you were testing VHDL templates. Now I see that you have been using Verilog. It probably doesn't make much sense to extend verismith with VHDL, because VHDL is a better language than Verilog. However, it would be very valuable to have a set of memory templates for testing. I think that @hackfin has been working on that.

@mithro, thanks for the references!

rodrigomelo9, 2020-03-27:
Mmm I got your point. The thing is, that I am downloading files without a license on them, without signing any kind of agreement (they are publicly available). Also, I am no producing a bitstream to use in a device. If I can open the files with any editor, why not to run synthesis? I know, is only my point of view.

> I thought that you were testing VHDL templates.
I was testing Verilog files with Yosys, and I will test the VHDL versions using ghdlsynt + Yosys.

About the set of memory templates written in VHDL, I have a bunch of them [here](https://github.com/INTI-CMNB-FPGA/fpga_lib/tree/master/vhdl/mems). There you can find a single-port RAM, a simple dual-port RAM and a true dual-port RAM, where using generics you can select the mode (read-first, write-first or no-change). You can use them (BSD 3-clasue), which were successfully implemented with ISE, Vivado, Quartus and Libero-Soc, and I can provide (develop) more if needed. Let me know @hackfin the status and what is still needed.

closed 2020-03-27 by tgingold

eine, 2020-03-27:
> The thing is, that I am downloading files without a license on them, without signing any kind of agreement (they are publicly available).

That's good. I'm used to having to sign agreements for downloading anything from Xilinx.

> where using generics you can select the mode (read-first, write-first or no-change).

Note that such sources hit two areas of GHDL. On the one hand, the frontend/analysis, where VHDL language features are handled. On the other hand, the backend/elaboration, where the inference is done. I think that the issue now is that GHDL + Yosys cannot handle some types of BRAM instantiation. It needs to be fixed before (or in parallel with) "fancy" language features which allow nicer and more compact descriptions.

Don't take me wrong. I think that the sources in your repo are very interesting, because you have tested them with a bunch of vendor tools. It is smaller and should be easier to test than [PoC](https://github.com/VLSI-EDA/PoC). Actually, I added it to ghdl/ghdl#974. However, you might (or might not) find issues not specifically related to RAM inference. Anyway, please do not hesitate to open issues for the bugs/erros you find.

> Let me know @hackfin the status and what is still needed.

See ghdl/ghdl#1069.

rodrigomelo9, 2020-03-27:
Don't worry, I will open issues if I found them.

There are several things to do everywhere, and is very difficult to track all of them :P I am particularly strong with VHDL (93) and GHDL used for simulation. I have also a good knowledge about vendors EDA tools (mainly ISE, Vivado, Quartus and Libero-SoC, for pure PL designs) and their Tcl implementations. I can also helps with bash scripts and Makefiles. Keep me in mind if the project needs something related.

eine, 2020-03-27:
> There are several things to do everywhere, and is very difficult to track all of them :P

Such a true thing 😄 

> I am particularly strong with VHDL (93) and GHDL used for simulation.

I would be glad to address any issues you might find with ghdl.rtfd.io. Specially having new users in mind. There are many ideas (https://ghdl.readthedocs.io/en/latest/development/Documentation.html) but little time. Any hint is welcome to prioritise. Contributions/PRs are also welcome, of course!

> I have also a good knowledge about vendors EDA tools (mainly ISE, Vivado, Quartus and Libero-SoC, for pure PL designs) and their Tcl implementations. I can also helps with bash scripts and Makefiles. Keep me in mind if the project needs something related.

Should you be willing to help, I think this is an area where the open source HDL community is most weak ATM the moment. There is [VUnit](https://github.com/vunit/vunit), which is a test runner written in Python and it includes multiple VHDL libraries and verification components for simulation. It also includes [OSVVM](https://osvvm.org/) as a dependency. VUnit supports several simulators (http://vunit.github.io/about.html#simulators), including GHDL. However, VUnit is mostly a test runner, not a project manager.

There are multiple project management tools around: [PoC](https://github.com/VLSI-EDA/PoC), [tsfpga](https://gitlab.com/truestream/tsfpga), [edalize](https://github.com/olofk/edalize), [fusesoc](https://github.com/olofk/fusesoc), [duh](https://github.com/sifive/duh)... (I think all of these support GHDL directly or indirectly) but there is no standard configuration file format. That's because, as you already know, there are lots and lots of variables to take into account. In the end, each project defines a declarative format and/or a DSL. Some are more complete/well-thought, others are naive... but the point is that there's a lot of duplicated effort.

Apart from simulation, synthesis and P&R, there is also an area with language server features. All of these require some common HDL sources (the design) and multiple complementary HDL/Python/JSON/YAML files. I created [vhdl-cfg](https://github.com/eine/vhdl-cfg) as a playground to explore similarities and differences between all these tools. The idea is to take simple examples (UART, adder, pipelined DSP, etc.) an use them with many different tools. Unfortunately, as you see, I've been side-tracked.

For example, I wrote https://github.com/eine/vhdl-cfg/blob/master/GHDLSynth.py in Python, because tsfpga, edalize and fusesoc are written in Python. However, it is based on the makefile from [antonblanchard/ghdl-yosys-blink](https://github.com/antonblanchard/ghdl-yosys-blink), which is (as said above) the base of the makefiles in the examples subdir of this repo.

As you said, lots of open tasks... So, just feel free to ask about whichever content is more appealing to you. I will make my best to guide you through existing content.

mithro, 2020-03-27:
(@eine If you haven't already, you should add these to my list at http://bit.ly/pyfgpalibs )

PR #97 (2020-03-27): Fixed URL of the ghdl/synth Docker image
Closes #95 

I also added a small notice about the automatic download of the image when used the first time (thinking in people not familiarized with Docker). Let me know if you prefer that I remove this notice.