ghdl/ghdl-yosys-plugin #99: Latches are unsupported

rodrigomelo9, 2020-04-02:
Hello. I am [analyzing with the plugin](https://github.com/rodrigomelo9/yosys-versus) (branch ghdl) examples provided by Xilinx. I found that latches are unsupported. Here only one of them as an example:

```
library ieee;
use ieee.std_logic_1164.all;

entity latches is
 port(
  G, D, CLR : in  std_logic;
  Q         : out std_logic
 );
end latches;

architecture archi of latches is
begin
 process(CLR, D, G)
 begin
  if (CLR = '1') then
   Q <= '0';
  elsif (G = '1') then
   Q <= D;
  end if;
 end process;
end archi;
```

```
$ $DOCKER_CMD ghdl/synth:beta yosys -Q -m ghdl -p "ghdl examples/vivado/latches.vhd -e latches"

-- Running command `ghdl examples/vivado/latches.vhd -e latches' --

1. Executing GHDL.
examples/vivado/latches.vhd:15:2:error: latch infered for net "q"
ERROR: vhdl import failed.
```

With ISE, it is only a warning:
```
WARNING:Xst:737 - Found 1-bit latch for signal <Q>. Latches may be generated from incomplete case or if statements. We do not recommend the use of latches in FPGA/CPLD designs, as they may lead to timing problems.
```

I know that they must be avoided, I never left designs with latches, but I think that must be supported by a synthesizer (with a warning of course). Some reasons could be that there are experiments that use them (ring oscillators) and for compatibility with existing designs.

If they are not supported by a design decision, the message could be "unsupported" and documented somewhere (unsupported known things in the README.md?).

Regards

Xiretza, 2020-04-02:
This isn't an issue with the yosys plugin, but rather ghdl itself - see ghdl/ghdl#938.

rodrigomelo9, 2020-04-02:
Hi Xiretza, thanks for the information. I used GHDL for years for simulation, but I am new with synthesis. I am using the plugin (because I was testing Yosys against ISE/Vivado in my repository).

Where I must post my issues? In the GHDL repository when the complaint is in the section *1. Executing GHDL.* of Yosys and here if other sections? Can I assume that the error in this first section of Yosys is the same that doing synthesis with GHDL? (I assume yes because it is solved by libghdl. Am I right?).

This info can be added into README.md as a notice and could be useful to have a location (probably in the ghdl repo, because is where `--synth` support is) with the section *Unsupported known features (to be implemented)* or something similar :P (if exists, please point me).

Regards

Xiretza, 2020-04-02:
As a rule of thumb: if `ghdl --synth ...` throws the same error as `yosys -p 'ghdl ...'`, then the issue is with the actual synthesis backend in ghdl. The yosys plugin really just translates between the ghdl and yosys intermediate representations, so most issues with the plugin are missing gate transformations shown as `Unsupported: instance \x of $gate` (see #96 for an example of that).

eine, 2020-04-02:
@rodrigomelo9, precisely, see section Issues of #98. The proposal was not accepted, and we still have separate issue trackers; hence the suggestion by @Xiretza applies. Nonetheless, many issues can be reported in either place and Tristan can move them.

> This info can be added into README.md as a notice and could be useful to have a location (probably in the ghdl repo, because is where --synth support is) with the section Unsupported known features (to be implemented) or something similar :P (if exists, please point me).

I believe that the expected location for this type of content is https://ghdl.readthedocs.io/en/latest/development/Synthesis.html. Sources are available in ghdl/ghdl: https://github.com/ghdl/ghdl/tree/master/doc/development.

Naturally, if this repo is going to be kept separate for the yosys plugin only, it would make sense to add a `doc` subdir in this repo. It would contain documentation/notes/explanations that are exclusive to the plugin and/or how to co-synthesise mixed-language projects using GHDL + yosys.

I can take care of setting up the publication of the `doc` in this repo, either as a separate site or merged into ghdl.rtfd.io.

tgingold, 2020-04-03:
At some points, there should be a -flatch option, or maybe just a warning for latches.

Is it important to you right now ?

rodrigomelo9, 2020-04-03:
No, is not important. In my opinion, a warning is enough. It can be closed because https://github.com/ghdl/ghdl/issues/938 exists. Regards

closed 2020-05-09 by tgingold