ghdl/ghdl-yosys-plugin #100: Instancing initialized RAMs (Verilog implementation)

hackfin, 2020-04-03:
Scenario: Instancing a RAM element:
```
init_ram_l:
	DPRAM16_init_hex_ce
	generic map ( ADDR_W => ADDR_W, DATA_W => DATA_W,
		INIT_HEX  => "test_a.hex" )
	port map (..)
```
from a Verilog implementation:
```
module DPRAM16_init_hex_ce #(
	parameter DATA_W = 16,
	parameter ADDR_W = 13,
	parameter INIT_HEX = "mem.hex",
	parameter SYN_RAMTYPE = "block_ram"
	
)
...
initial begin
	$readmemh(INIT_HEX, mem);
end
...
```
Result:
`ERROR: Can not open file `mem.hex` for \$readmemh.`

Expected: Should pass "test_a.hex" to `$readmemh()`

Also, instancing several DPRAM16_init_hex_ce instances will fail.
Submitting test case shortly.

tgingold, 2020-04-06:
Are there two issues (file to be open and multiple definition) ?

hackfin, 2020-04-07:
I should have commented better, there are 2 cases to be tested, but it's in fact two cascaded errors from `INIT_HEX` parameter not replacing the default `mem.hex`
When creating `mem.hex` anyway (`touch mem.hex`), it seems to want to create another module with the same signature. So for the case if someone wants to instance several identical modules with the same initialization values, there should probably be an extra test, if first issue is solved.

tgingold, 2020-04-07:
For: Expected: Should pass "test_a.hex" to $readmemh()

That part is correct.  If you dump the AST before importing the verilog file, INIT_HEX is correctly set.
You can check that: if you just read the vhdl files and write the output (write_ilang or write_verilog), the parameter is correctly set.
Also, if you simply read the verilog file, it fails (trying to read `mem.hex`).

When you read the verilog file, it tries to directly synthesize it.  So you need to read the verilog file using the `-defer` option.  Then you need to elaborate the verilog module (using `hierarchy` ?)

hackfin, 2020-04-09:
My bad, I wasn't aware of the -defer option. Havent verified, but first case resolves now. Haven't found out yet about the dual case (see `testsuite.sh`)

tgingold, 2020-04-09:
I have just pushed a tiny commit to handle that case.

hackfin, 2020-04-09:
Namespaces resolve now, however for both cases, `DPRAM16_init_hex_ce` map any further to `$mem*` cells because they're tagged as blackbox.
Check:
```
run_yosys -p "ghdl top.vhdl ../../../library/libram.vhdl -e top;
	read_verilog -defer ../../../library/verilog/dpram16_init.v;
	hierarchy; synth_ecp5; stat; write_ilang dpram_single.il"
```
Stats show:
```
     CCU2C                           4
     DPRAM16_init_hex_ce             1
     TRELLIS_FF                      8
```
Checking ilang output, `INIT_HEX` is indeed properly set, but on the way, no `$mem/$memrd/$memwr` is generated. Not sure if it's a yosys issue.

tgingold, 2020-04-09:
I suppose the plugin code has to instantiate the verilog module.  That's something I need to investigate.

closed 2020-06-29 by hackfin