ghdl/ghdl-yosys-plugin #101: ERROR: Assert `is_fully_const() && GetSize(chunks_) <= 1' failed in kernel/rtlil.cc

rodrigomelo9, 2020-04-05:
I was [verifying GHDL `--synth` and the `ghdl-yosys-plugin` with examples from Xilinx](https://github.com/rodrigomelo9/verifying-foss-hdl-synthesizers). I found a problem with a file for ISE, called **counters_3.vhd**, which fails only with the plugin:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_unsigned.all;

entity counters_3 is
    port(C, ALOAD : in std_logic;
         D : in std_logic_vector(3 downto 0);
         Q : out std_logic_vector(3 downto 0));
end counters_3;

architecture archi of counters_3 is
    signal tmp: std_logic_vector(3 downto 0);
begin
    process (C, ALOAD, D)
    begin
        if (ALOAD='1') then
            tmp <= D;
        elsif (C'event and C='1') then
            tmp <= tmp + 1;
        end if;
    end process;

    Q <= tmp;

end archi;
```

```
$ docker run --rm -t -v $(pwd):/src -w /src ghdl/synth:beta yosys -m ghdl -Q -p 'ghdl -fsynopsys examples/ise/HDL_Coding_Techniques/counters/counters_3.vhd -e counters_3'

-- Running command `ghdl -fsynopsys examples/ise/HDL_Coding_Techniques/counters/counters_3.vhd -e counters_3' --

1. Executing GHDL.
Importing module counters_3.
ERROR: Assert `is_fully_const() && GetSize(chunks_) <= 1' failed in kernel/rtlil.cc:3785.
```

tgingold, 2020-04-06:
The problem is that dff with non-constant reset value is not supported by yosys.

So, maybe you'd prefer a better error message.

rodrigomelo9, 2020-04-06:
I see... Now that you said that, I realized that I reported an issue in Yosys, on the Verilog version  (https://github.com/YosysHQ/yosys/issues/1809). It was synthesized using Yosys without complaints. The problem was present with `ngdbuild` (ISE). I am not sure about what is better, easier, convenient...
* Abort using the message "unsupported by Yosys"?
* Do synthesize anyway and left the complaint in the next step?
* Is there another alternative? The current message is somewhat cryptic.

tgingold, 2020-04-07:
In fact, there is support of dffsr.  I will modify the code for it.  Will push soon.

Thank you for the issue.

closed 2020-04-07 by tgingold