ghdl/ghdl-yosys-plugin #104: case statement 'when x to y' causes a "GHDL Bug Occured"

pytricke, 2020-04-15:
Using the ghdl-yosys plugin a "GHDL Bug Occured" is reported when using a 'when x to y' statement (inside a case statement).

It can be reconstructed by saving the code below (testcase1.vhd) and running:
$> ghdl -a testcase1.vhd
$> yosys -m ghdl -p 'ghdl 'testcase1.vhd' -e 'testcase1'; synth -top 'testcase1';'

The 'normal' GHDL compiler does not complain about this kind of construct, but the plugin seems to raise an exception. The ouput is listed at the end of this bug report.

-- start testcase1.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity testcase1 is
   port (
          sel   : in  unsigned(1 downto 0);
          det   : out std_logic
        );
end testcase1;

architecture behavior of testcase1 is
begin

tc: process(sel)
    begin
       case to_integer(sel) is
          when 0 to 1 =>
             det <= '0';
          when others =>
             det <= '1';
       end case;
    end process;

end behavior;
-- end testcase1.vhd

-- start output
 Yosys 0.9+2406 (git sha1 5d18fdad, clang 7.0.1-8 -fPIC -Os)
-- Running command `ghdl testcase1.vhd -e testcase1; synth -top testcase1;' --
1. Executing GHDL.
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-243-g921e989f) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
Command line:
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-stmts.adb:912
******************************************************************
ERROR: vhdl import failed.
-- end output

Xiretza, 2020-04-15:
Formatted for better readability:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity testcase1 is
    port (
        sel : in unsigned(1 downto 0);
        det : out std_logic
    );
end testcase1;

architecture behavior of testcase1 is
begin
    tc: process(sel)
    begin
        case to_integer(sel) is
            when 0 to 1 =>
                det <= '0';
            when others =>
                det <= '1';
        end case;
    end process;
end behavior;
```

`ghdl --synth testcase1.vhd -e testcase1` fails with the same error, so this isn't related to the plugin:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-320-g90d7bfe9) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
/tmp/
Command line:
ghdl --synth testcase1.vhd -e testcase1
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-stmts.adb:918
Call stack traceback locations:
0x52df5e 0x53058b 0x52f668 0x527e1b 0x52372a 0x56a143 0x4dce29 0x406e51 0x7fd4e09fe021 0x406ecc 0xfffffffffffffffe
******************************************************************
```

tgingold, 2020-04-15:
Ranges in case statements (like `when 0 to 1 =>`) are not yet handled.

pytricke, 2020-04-15:
Thank you Xiretza and tgingold for your quick replies.

tgingold, 2020-05-09:
Now implemented.  Thank you for the issue.

closed 2020-05-09 by tgingold

pytricke, 2020-05-11:
Thank you for implementing this useful feature!