ghdl/ghdl-yosys-plugin #107: Build break with yosys current repo head

donnie-j, 2020-04-22:
As of April 22, It seems the exported interface from yosys has changed:

jeff@JeffNB:~/work/tools/ghdl-build$ git clone ../ghdlsynth-beta/
Cloning into 'ghdlsynth-beta'...
done.
jeff@JeffNB:~/work/tools/ghdl-build$ 
jeff@JeffNB:~/work/tools/ghdl-build$ cd ghdlsynth-beta/
jeff@JeffNB:~/work/tools/ghdl-build/ghdlsynth-beta$ 
jeff@JeffNB:~/work/tools/ghdl-build/ghdlsynth-beta$ make
yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I/home/jeff/work/tools/usr/include -O
src/ghdl.cc: In function ‘void import_module(Yosys::RTLIL::Design*, GhdlSynth::Module)’:
src/ghdl.cc:570:29: error: ‘class Yosys::hashlib::idict<Yosys::RTLIL::IdString>’ has no member named ‘insert’
    module->avail_parameters.insert(to_str(get_param_name(m, idx)));
                             ^~~~~~
Makefile:25: recipe for target 'ghdl.o' failed
make: *** [ghdl.o] Error 1
jeff@JeffNB:~/work/tools/ghdl-build/ghdlsynth-beta$

Thanks again, very exciting to see GHDL synth advance so quickly,

tgingold, 2020-04-23:
Now fixed.

closed 2020-04-23 by tgingold

tmeissner, 2020-04-23:
I still get a similar error with current master of ghdl-yosys-plugin:

```bash
Step 5/9 : RUN cd /root &&     mkdir ghdl-yosys-plugin &&     cd ghdl-yosys-plugin &&     curl https://codeload.github.com/ghdl/ghdl-yosys-plugin/tar.gz/master | tar xzf - --strip-components=1 &&     make GHDL=/opt/ghdl/bin/ghdl YOSYS_CONFIG=/opt/yosys/bin/yosys-config &&     make install GHDL=/opt/ghdl/bin/ghdl YOSYS_CONFIG=/opt/yosys/bin/yosys-config
 ---> Running in 93e4d71847b5
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 62553    0 62553    0     0  88980      0 --:--:-- --:--:-- --:--:-- 88853
/opt/yosys/bin/yosys-config --exec --cxx -c --cxxflags -o ghdl.o src/ghdl.cc -fPIC -DYOSYS_ENABLE_GHDL -I/opt/ghdl/include -O
src/ghdl.cc:570:29: error: no member named 'insert' in 'Yosys::hashlib::idict<Yosys::RTLIL::IdString, 0, Yosys::hashlib::hash_ops<Yosys::RTLIL::IdString> >'
                        module->avail_parameters.insert(to_str(get_param_name(m, idx)));
                        ~~~~~~~~~~~~~~~~~~~~~~~~ ^
1 error generated.
make: *** [Makefile:25: ghdl.o] Error 1
The command '/bin/sh -c cd /root &&     mkdir ghdl-yosys-plugin &&     cd ghdl-yosys-plugin &&     curl https://codeload.github.com/ghdl/ghdl-yosys-plugin/tar.gz/master | tar xzf - --strip-components=1 &&     make GHDL=/opt/ghdl/bin/ghdl YOSYS_CONFIG=/opt/yosys/bin/yosys-config &&     make install GHDL=/opt/ghdl/bin/ghdl YOSYS_CONFIG=/opt/yosys/bin/yosys-config' returned a non-zero code: 2
Makefile:25: recipe for target 'ghdl-formal' failed
```

commit 0a8a2e95 (2020-04-23): ghdl.cc: adjust for recent yosys.  Fix #107