ghdl/ghdl-yosys-plugin #108: Cannot handle dynamic slicing and other operations in a single statement

kammoh, 2020-04-22:
MWE:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity MWE is

  port (
    clk : in std_logic;
    rst : in std_logic;
    a   : in std_logic_vector(0 to 8 - 1);
    b   : in std_logic_vector(0 to 8 - 1);
    o   : out std_logic_vector(0 to 2 - 1)
  );
end MWE;

architecture behavioral of MWE is

  signal cnt : integer range 0 to 4 - 1;
begin
  -- counters : process (clk)
  -- begin
  --   if rising_edge(clk) then
  --     if rst = '1' then
  --       cnt <= 0;
  --     else
  --       cnt <= cnt + 1;
  --     end if;
  --   end if;
  -- end process counters;
  o <= a(cnt*2 to (cnt + 1)*2 - 1) xor b(cnt*2  to (cnt + 1)*2 - 1);
end architecture behavioral;
```

works fine in simulation.
In synthesis:
```
$ yosys -m ghdl -p 'ghdl mwe.vhdl -e MWE' 

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-371-gb800347e) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-apple-darwin15
/Volumes/src/ghdl-yosys-mwe-3/
Command line:

Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-oper.adb:131 discriminant check failed
******************************************************************
ERROR: vhdl import failed.
```

tgingold, 2020-04-23:
Now fixed.

closed 2020-04-23 by tgingold

kammoh, 2020-04-23:
While the trimmed down MWE works now, it still fails other instances like this one:

```vhdl
 tag_match <= '1' when ((c_dout_r(word_cnt_r*CCW to (word_cnt_r+1)*CCW -1) xor bdo_offset_mux) = bdi) else '0';
```

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-386-g5b9fb819) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-apple-darwin15
/Volumes/src/lwc/lwc_candidates_rtl/Pyjamask/src_rtl/
Command line:

Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-oper.adb:131 discriminant check failed
******************************************************************
ERROR: vhdl import failed.
```

As before, no information on which file / line / compilation unit / entity this is happening, so have to do a binary search with commenting out code to find where it actually chokes :))

kammoh, 2020-04-23:
Well, it does not seem to even pass the MWE
I made sure I had the latest `git` and rebuilt and reinstall both `ghdl (d8cd7bc8)` and `ghdl-yosys-plugin (386ad815)`. Which PR corresponds to the fix?

tgingold, 2020-04-24:
> GHDL release: 1.0-dev (v0.37.0-386-g5b9fb819) [Dunoon edition]

It looks like you are using a version of ghdl before the fix was pushed.
Otherwise, an mwe would be nice.

The crash is really unexpected.  You can use '-v' to view which instance was synthesized if you want to trim down.
But that's not required.  I will be happy with a large file as I will easily know which construct was the cause of the crash.

An MWE must be standalone: the reproducer must not require external modules or libraries.  Having a small MWE is nice as that's easier to handle, but not really necessary.