ghdl/ghdl-yosys-plugin #112: Feature suggestion: add a `ghdl_write_vhdl` command to yosys

rlee287, 2020-04-22:
This would greatly help with simulating mixed-language projects, which could look like the following:

Verilog testbench (which should be possible now):
1. Use yosys to synthesize the modules (which may be a mix of Verilog and VHDL sources)
2. `write_verilog` the top module.
3. Use Icarus Verilog with the Veriog testbench, which instantiates the device under test from the output of the yosys synthesis.

VHDL testbench (which would be made possible by the command suggested above):
1. Use yosys to synthesize the modules (which may be a mix of Verilog and VHDL sources)
2. `ghdl_write_vhdl` the top module.
3. Use GHDL with the VHDL testbench, which instantiates the device under test from the output of the yosys synthesis.

tgingold, 2020-04-23:
Of course, that's something that would be great.
As it doesn't require any knowledge of ghdl, I think this could be a contribution from anyone interested.  Just start from the existing write_verilog command of yosys.

rlee287, 2020-05-20:
I am going through the `write_verilog` file now to see how it is structured and what kinds of changes would be necessary. However, it also looks like `ghdl --synth` outputs VHDL to the standard output. Would it make more sense to try to repurpose some of the code from `ghdl --synth`, or would it be better to work directly with RTLIL the way `write_verilog` does?

tgingold, 2020-05-21:
The code from ghdl that writes the vhdl netlist is written in Ada, so you cannot repurpose it without rewriting it.
But you could get inspiration from it!

The most difficult issue is how to deal with identifiers.  The names in the netlist can be invalid vhdl names (like all the ones that start with $), can be vhdl keywords or can be correct identifiers.  An easy way to deal with this issue is to always use extended identifiers. That would be a first good step.

The second issue is to deal with sign extension.  The yosys operators can implicitly extended their operands, and that should become explicit in vhdl.

The third issue is to deal with some particularities.  In vhdl, there are functions to add signed or unsigned vectors, but not for bits.

Xiretza, 2020-05-21:
The code that turns VHDL into a netlist is also written in Ada, so shouldn't it be possible to create a bridge in the other direction as well?

tgingold, 2020-05-21:
Yes, but that would be a different project: adding an importer from yosys netlist to the ghdl netlist.

Possible, my in my opinion less useful or more difficult to use.