ghdl/ghdl-yosys-plugin #113: Make not working

anhangaba, 2020-04-28:
When running `make` I'm getting the following error message:

```
src/ghdl.cc: In function ‘void import_module(Yosys::RTLIL::Design*, GhdlSynth::Module)’:
src/ghdl.cc:570:59: error: no match for call to ‘(Yosys::hashlib::pool<Yosys::RTLIL::IdString>) (std::string)’
  570 |    module->avail_parameters(to_str(get_param_name(m, idx)));
```
When I switch to `81bb434` the make works as expected.

I'm using the latest, at the time of this writing, yosys (`a66200ed`) and ghdl (`e7972bd2`).

The `$PREFIX` for both ghdl and yosys is `/home/anhangaba/.local`.

tgingold, 2020-04-28:
I cannot reproduce the error (but I don't use the latest of yosys).  Is it a very recent error due to a change in yosys, or is it due to a different compiler ?

Xiretza, 2020-04-28:
Are you sure you're using latest yosys (did you `make install`)? This error happens with yosys headers (namely `kernel/rtlil.h`) before YosysHQ/yosys@06a344ef, but ghdl-yosys-plugin after 0a8a2e9 (which addressed this change in yosys, but is thus no longer compatible with older yosys).

anhangaba, 2020-04-28:
You are right, my yosys was actually `a66200ed`. The issue was my `DEST_DIR` env var was set to directory not in `$PATH`.

It works with the correct version of yosys.

closed 2020-04-28 by anhangaba