ghdl/ghdl-yosys-plugin #115: Incorrect "overflow check failed" raised

pytricke, 2020-05-06:
Please consider the following (simplified) testcase:

```VHDL

library ieee;
use ieee.std_logic_1164.all;

entity testcase is
   port (
          din    : in  std_logic_vector(3 downto 0);
          dout   : out std_logic_vector(1 downto 0)
        );
end testcase;

architecture behavior of testcase is
   signal testidx : natural range 3 downto 2;
begin

--------------------------------------------------------------
-- tc0 does not cause an overflow error
--tc0: process(din)
--     begin
--        if (din(3)='1') then
--           dout <= din(2 downto 1);
--        else
--           dout <= din(1 downto 0);
--        end if;
--     end process;
--------------------------------------------------------------

--------------------------------------------------------------
-- tc1 with the dout assignment does cause an overflow error
tc1: process(din)
     begin
        if (din(3)='1') then
           testidx <= 3;
        else
           testidx <= 2;
        end if;
     end process;

   dout   <= din(testidx-1 downto testidx-dout'length);
--------------------------------------------------------------

end behavior;

```

When running _ghdl --synth testcase.vhd -e testcase_
The following is output:
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-expr.adb:1253 overflow check failed
Call stack traceback locations:
0x555bae5628c6 0x555bae563541 0x555bae53c47c 0x555bae564fb2 0x555bae53b972 0x555bae53e5b2 0x555bae54adf3 0x555bae54b069 0x555bae53af4a 0x555bae53b09c 0x555bae534252 0x555bae59aba3 0x555bae4d9b91 0x555bae645761 0x555bae2999c3 0x7f92d898d0b1 0x555bae2981ac 0xfffffffffffffffe
******************************************************************

The _testidx_ signal is constrained to _3 downto 2_, but I suspect that this is not correctly 'evaluated' causing the _testidx-dout'length_ to result in the failed overflow check.

tgingold, 2020-05-06:
Now fixed.  Thank you for the issue.

closed 2020-05-06 by tgingold

pytricke, 2020-05-07:
Thank you for fixing it and for fixing it so quickly!