ghdl/ghdl-yosys-plugin #116: Generic edge causes "ill-formed clock-level" during synthesis

pytricke, 2020-05-11:
Please consider the following (simplified) testcase:

```VHDL
library ieee;
use ieee.std_logic_1164.all;

entity testcase3 is
   generic (
             edge    : std_logic := '1'
           );
   port (
          clk    : in  std_logic;
          D      : in  std_logic;
          Q      : out std_logic
        );
end testcase3;

architecture behavior of testcase3 is
begin

tc3: process(clk)
     begin
        if (clk'event and clk=edge) then
           Q    <= D;
        end if;
     end process;

end behavior;
```

During simulation GHDL does not complain about this construct, but when running:
_ghdl --synth testcase3.vhd -e testcase3_
the folllowing error is reported:
_testcase3.vhd:20:30:error: ill-formed clock-level, '0' or '1' expected_

tmeissner, 2020-05-11:
This can be moved to GHDL issues I assume, because you get the error when using GHDL, not zhe plugin.

pytricke, 2020-05-11:
Yes, I am very sorry if this was submitted in the wrong place. Is there an easy way for me to move it?

closed 2020-05-11 by pytricke

tmeissner, 2020-05-11:
Leave it open, @tgingold isn't that pedantic. Maybe @tgingold can move the bug.

tmeissner, 2020-05-11:
> Yes, I am very sorry if this was submitted in the wrong place. Is there an easy way for me to move it?

No reason to be sorry, @tgingold isn't that pedantic. Simple rule is, all errors which occur when using `ghdl --synth` are GHDL related, for most of the errors using GHDL in Yosys the right place is the ghdl-yosys-plugin.